FabMo/FabMo-Engine #490: Pocketing in CR command doesn't always remove the last bit of material in the center

wlyoung, 2016-07-28:
With the outside-in pocket parameter set, depending on the overlap and and size of the pocket there is sometimes a little bit of material left in the center.

gordonab, 2016-08-08:
Bill, can you give me an example of a pocket that doesn't finish? I just ran several dozen in preview that look like they go all the way to the center or really close. If not, I'll keep trying to break it.

wlyoung, 2016-08-08:
I'l see if I can duplicate the settings that left the little square in the
middle and report back

gordonab, 2016-08-08:
Thanks, that will help a bunch.

What kind of bit were you using?

wlyoung, 2016-08-08:
> issuecomment-238374434

1/4". It was in a parametric pocketing sbp file I was working on so it's
easy to duplicate...just need to figure out what the stepover setting was
in the VC command that left that little bit part uncut. I'll work on it
tomorrow

ryansturmer, 2016-08-30:
Any movement on this @gordonab @wlyoung ?

ryansturmer, 2016-09-02:
Moving this to the next release, so we can get this out tha door.

gordonab, 2016-09-03:
Working on this now. Should say, testing to see what parameters are causing
this situation.

ryansturmer, 2017-04-21:
@gordonab @wlyoung How are we doing on this one?  Can I close this out?  Or is there still work to do here?