FabMo/FabMo-Engine #491: Cannot create SBP jobs through the editor

Alex-Canales, 2016-07-29:
In the Editor, chose OpenSBP. Put:

    SA
    MH
    M2 1, 1

``Submit as Job``, save the file as ``editor.sbp``. In job manager, run it: the job manager shows it as "running job" but in fact launches the next job. Impossible to remove the job as "running job", need to kill the engine.

Logs when trying to run the job without another one pending:

    info: Running the next job in the queue [routes]
    info: Arming the machine for nextJob [machine]
    info: Firing automatically since authorization is disabled. [machine]
    info: Machine is deauthorized. [machine]
    info: Running next job [machine]
    info: printing[object Object] [db]
    info: 158 [db]
    info: [object Object] [machine]
    info: Running job {"file_id":7,"name":"OpenSBP (Editor)","description":"An OpenSBP job from the editor","created_at":1469801581305,"started_at":1469801585616,"finished_at":null,"state":"running","order":null,"_id":158} [machine]
    info: Running file /opt/fabmo/files/ad4fa410-5593-11e6-b179-f3de04960557.nc [machine]
    info: Got a machine state change: idle [machine]
    info: Connected G-Code Runtime [gcode]
    debug: Saving config to /opt/fabmo/config/instance.json [config]
    error: 102,STAT_BAD_NUMBER_FORMAT,number format error [machine]
    error: 102,STAT_BAD_NUMBER_FORMAT,number format error [machine]
    error: 132,STAT_MCODE_COMMAND_UNSUPPORTED,M command is not supported [machine]
    debug: GET /jobs/queue [engine]
    debug: GET /jobs/history?start=0&count=10 [engine]
    debug: fsync()ed /opt/fabmo/config/instance.json [config]

Note: there is no issue when submitting a SBP file. The issue is only when the file is created by the editor.

ryansturmer, 2016-08-08:
Are you selecting OpenSBP as the language when you are using the editor?  Does the syntax hilighting for the job look normal?

[image]

Alex-Canales, 2016-08-08:
Yes, I did select OpenSBP, the syntax looked normal.

ryansturmer, 2016-08-30:
@Alex-Canales this is definitely the job being submitted as a g-code file instead of an SBP file somehow.  Can you provide specific steps to reproduce?

ryansturmer, 2016-09-02:
This appears not to be the case in the latest RC.  I am going to close it for now, and we can re-open in the event that it turns up again.

closed 2016-09-02 by ryansturmer