FabMo/FabMo-Engine #492: Job manager: need to double click on buttons each time an action is triggered

Alex-Canales, 2016-07-29:
In the Job Manager, when I perform an action (like pending a job in the queue or delete a job in the history), if I want to click on another button, I have to click twice.

Before the first click, the cursor is not displayed as a pointer (the hand) when "hovering" a button. After the first click, the behaviour is back to normal.

I do not think this issue was there in July, 13th 2016.

ryansturmer, 2016-08-08:
Alex - I assume you're observing this on firefox, under linux?  Can you post your version?

-R

Alex-Canales, 2016-08-08:
On ArchLinux.
Mozilla Firefox 47.0.1

SW0rden, 2016-08-09:
This happens on a windows PC/desktop in Chrome also.

In addition to this, the latest version of the job manager now shows horizontal/vertical scrollbars and when you click anywhere in the up next frame, even when clicking on the play/run arrow, the up next area shrinks down as if you are going to drag/drop/rearrange.  I wouldn't expect it to behave this way when clicking the play/run arrow.  Recent jobs also have vertical scrollbars.

Would a  'remove from list' of the recent jobs be useful?  This would then bring up the next in line.  Not sure, need to play with that more.

closed 2016-09-02 by ryansturmer