FabMo/FabMo-Engine #494: Locked out of FABMO after updating to new password-protected version 1.2.0

stevempotter, 2016-07-30:
I am halfway through updating my FabMo for Handibot2.
I think I successfully updated to v.1.2.0 from v.1.1.0 because I can no longer access my Handibot on 192.168.42.1.  I just keep getting "Connection refused" on any browser I have tried.  I have not yet seen any login windows popping up, in which I should be able to enter the default password and user and then go in to personalize it. How can I get to the login window? I do not have pop-ups blocked.
Thanks,
Steve

tedh-shopbot, 2016-07-30:
So Steve, give me a little more info ... what device are you trying to access from? Is this an update you were attempting following the instructions in the email to Handibotters, or directly from the Configuration panel? Did the Handibot have an internet connection at the time your started?

If you look at your wifi manager on your device, do you see any available networks/devices with the name starting FEZ....?

=Ted

stevempotter, 2016-07-30:
Hi Ted,

Thanks, I will try to clarify. I was following the attached update guide
pdf.
Yes, I had an internet connection to the Handibot when I started. I
successfully updated the Updater. Then I think I successfully updated the
Software to the latest development release. I did before all this, and
still do see the FabMo-FZED... SSID and can connect to it with my Macbook
Air (and with my android phone). It is only when I try to get the FabMo web
page up by going to 192.168.42.1 do I get "Connection refused"  (see
attached screenshot). This is not "not found" but a refused error, which I
assume is a result of not entering the right password. That is because
no window ever pops up to ask me for my login info. I tried several
browsers and do not have pop ups blocked.

Thanks for your help,
Steve

________________________________________________
We are now living in Ireland... Until we get phones sorted, continue to
contact me by leaving voicemail at +1 (678) 995 5067 (my Google
Voice/Hangout number)

tedh-shopbot, 2016-07-30:
Hi Steve,

Sorry the update seems to have not quite made it ... usually they won't be so messy. Anyway, I believe we have a fix for you. I've sent a dropbox link to your gmail address so that you can download an FMU file.

- While you still have an internet connections, download the FMU file to the device you are working with. The Macbook should be fine ...

- Then, connect to the Handibot in direct (AP) mode using your wireless manager in the usual manner.

- Next, open 192.168.42.1:81    ... the :81 will get you to the "Updater" which we believe should be running. 

- In the left-side menu, select FMU Update and click the green update button to browse for the FMU file you downloaded. Running it should take care of the problem.

- Just for kicks, let's also run the Firmware update from the left-side menu. This should go quickly, and after it has finished, you should see a blue light on the control card pulsing about every 4 sec ... the g2 heartbeat.

- Now, clicking the "Go to Dashboard" button should take you back to the normal FabMo interface to your Handibot. And, at this point you will probably encounter the new login.   admin     go2fabmo

Hope that gets you going. If not, let us know.

=Ted

stevempotter, 2016-07-31:
Thanks, Ted. But I did not get any link from dropbox for the fmu file. Did
you share it with
steve.potter@gmail.com ?  I searched all my mailboxes, not there.

Please try again or send it directly to me via gmail if it is less than
20MB.
I look forward to finishing the update soon, with your helpful instructions!

-Steve

________________________________________________
We are now living in Ireland... Until we get phones sorted, continue to
contact me by leaving voicemail at +1 (678) 995 5067 (my Google
Voice/Hangout number)

stevempotter, 2016-07-31:
Hi Ted,

Since I don't have a link to the new .fmu on Dropbox yet, I tried
re-updating the using "production_4_19_2016.fmu" the file I have. Getting
to the Updater worked as per your instructions. Just FYI, that did not fix
the problem, I am still locked out of the Dashboard. Connection refused, no
login window popping up. Can't use my Handibot on this 3-day weekend here
in Ireland :(

Please send the fixed .fmu.

Thanks,
Steve

________________________________________________
We are now living in Ireland... Until we get phones sorted, continue to
contact me by leaving voicemail at +1 (678) 995 5067 (my Google
Voice/Hangout number)

tedh-shopbot, 2016-07-31:
Hi Steve,
Per the email you should have received from Brian at the gmail address you sent. You should have received the files ... or at least should now have access via the dropbox.
I've also appended the instructions in a recent email ...
=Ted

closed 2016-07-31 by tedh-shopbot

stevempotter, 2016-07-31:
Thanks Ted, I got the new .fmu and tried the fmu update in AP mode as
instructed. It threw an error saying it can't do update in AP mode:

___________________

Updater Console

info: Client 192.168.42.20 connected. [websocket]

shell:  [hooks]

info: Creating upload 2384b3c0-00ec-11e6-acbe-335348225d72 [routes]

info: Recieved file #0 (full-update-edison-7_18_2016.fmu) for upload
2384b3c0-00ec-11e6-acbe-335348225d72 [routes]

info: Upload of 2384b3c0-00ec-11e6-acbe-335348225d72 complete. [routes]

info: Upload complete [routes]

info: Doing FMU [routes]

info: Starting task: 2398b0f0-00ec-11e6-acbe-335348225d72 [updater]

shell: Creating fmu directory...

       Entering fmu directory...

       Unpacking FMU: /tmp/upload_0a906126350c7da1602abf75520faeb3

       install

       files/fmu-update.sh

       files/fmu-update.service

       Applying update...

       Checking the environment...

       !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

          Cannot perform this update from AP mode!

       !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

        [hooks]

info: Finishing task 2398b0f0-00ec-11e6-acbe-335348225d72 with a state of
failed [updater]

error: Did not execute FMU successfully. [hooks]

info: Client 192.168.42.20 connected. [websocket]
____________

What shall I try next?

Thanks,
Steve
________________________________________________
We are now living in Ireland... Until we get phones sorted, continue to
contact me by leaving voicemail at +1 (678) 995 5067 (my Google
Voice/Hangout number)

tedh-shopbot, 2016-07-31:
Arrgh … My bad.
For the update fmu, you will need to be connected to the internet.
-so … reconnect in direct (AP) mode to 192.168.42.1:81
-use the network manager on the left side panel of the updater; it will put the Updater on the network let you log on (but you will need the old network IP, followed by :81) … {{hopefully if you don’t know it, you’ve got Fing on your Android and can determine it}} … you should now have internet
-now try running the update FMU from the folder
-**It will disconnect and THEN take about 10 minutes to update all the stuff it needs … before it restarts; at that point a refresh should bring it back
-then firmware
-then do the production FMU … or you can go check on the FabMo dashboard and come back for this update.

I had forgotten that the production FMU are complete and you do not need to be connected, that update FMU goes to the web for downloads.

=Ted

tedh-shopbot, 2016-08-01:
Hi Steve,

Can you confirm that you ran the firmware update, after your recent attempt at updating and that there is the 4sec blue heartbeat on the card?

=Ted

PS. I closed the Github issue so that I could focus on using email where we appear to be able to send files. So I did not look back there until just now. If you want to switch back there it is ok, but let me know as I am trying to keep up mobile.