FabMo/FabMo-Engine #495: Still locked out of FabMo

stevempotter, 2016-07-31:
Thanks Ted, but the issue was closed prematurely.

I got the new .fmu and tried the fmu update in AP mode as
instructed. It threw an error saying it can't do update in AP mode:
…

___________________

Updater Console

info: Client 192.168.42.20 connected. [websocket]

shell:  [hooks]

info: Creating upload 2384b3c0-00ec-11e6-acbe-335348225d72 [routes]

info: Recieved file #0 (full-update-edison-7_18_2016.fmu) for upload
2384b3c0-00ec-11e6-acbe-335348225d72 [routes]

info: Upload of 2384b3c0-00ec-11e6-acbe-335348225d72 complete. [routes]

info: Upload complete [routes]

info: Doing FMU [routes]

info: Starting task: 2398b0f0-00ec-11e6-acbe-335348225d72 [updater]

shell: Creating fmu directory...

       Entering fmu directory...

       Unpacking FMU: /tmp/upload_0a906126350c7da1602abf75520faeb3

       install

       files/fmu-update.sh

       files/fmu-update.service

       Applying update...

       Checking the environment...

       !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

          Cannot perform this update from AP mode!

       !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

        [hooks]

info: Finishing task 2398b0f0-00ec-11e6-acbe-335348225d72 with a state of
failed [updater]

error: Did not execute FMU successfully. [hooks]

info: Client 192.168.42.20 connected. [websocket]

____________

What shall I try next?

Thanks,
Steve

stevempotter, 2016-07-31:
Hi Ted,

Here is something I tried but still did not fix my problem:
I used the Updater console to connect to my Wifi network so the Handibot was not in AP mode. Then I found out its IP address and reconnected to the Updater on socket 81, and tried the .fmu update again.
This time I did not get an error message saying it can't do the update in AP mode (since I was no longer in AP mode). I soon got the "Disconnected" window popping up. I waited 15 minutes, power-cycled the Handibot, but still I get the "Connection Refused" message and no login window when I select "go to dashboard" from the updater:
____
This site can’t be reached
192.168.8.104 refused to connect.
Try:
Reloading the page
Checking the connection
Checking the proxy and the firewall
ERR_CONNECTION_REFUSED
____

Nor can I get to the Dashboard when I turn off my home's Wifi and connect to the Handibot in AP mode. Connection refused, no login windows.  Refreshing the page did not help.

Other ideas?
-
Steve

ryansturmer, 2016-08-08:
Steve - is there a blinking blue LED on the back of your control card?  You should be able to see it through the clear acrylic on the back of your machine.

closed 2016-12-08 by ryansturmer