FabMo/FabMo-Engine #496: OpenSBP: Cannot view code when using MH command

Alex-Canales, 2016-08-02:
It is possible to execute code using MH command but not displaying it.

Here is the code used:

    SA
    MH
    M2 1, 1

Here is the log error when trying to view it:

    info: Got a machine state change: idle [machine]
    info: Starting OpenSBP program [sbp]
    warn: More parameters passed into SA (1) than are supported by the command. (0) [sbp]
    warn: Attempting to execute a stack-breaking command without G2, which may cause issues. [sbp]
    warn: More parameters passed into MH (1) than are supported by the command. (0) [sbp]
    error: There was a problem executing a stack-breaking command:  [sbp]
    error: TypeError: Cannot read property 'driver' of null
        at SBPRuntime.exports.MH (/home/alex/Shopbot/Engine/runtime/opensbp/commands/move.js:171:14)
        at SBPRuntime._executeCommand (/home/alex/Shopbot/Engine/runtime/opensbp/opensbp.js:778:5)
        at SBPRuntime._execute (/home/alex/Shopbot/Engine/runtime/opensbp/opensbp.js:844:16)
        at SBPRuntime._continue (/home/alex/Shopbot/Engine/runtime/opensbp/opensbp.js:482:12)
        at runCallback (timers.js:570:20)
        at tryOnImmediate (timers.js:550:5)
        at processImmediate [as _immediateCallback] (timers.js:529:5) [sbp]
    info: End of program reached. (pc = 4) [sbp]
    info: No g-codes remain to dispatch. [sbp]

Maybe @gordonab could explain me why this is working when executed but not when viewed.

gordonab, 2016-08-02:
Hi Alex,

I'm running this on an Edison and not getting the error. I'll try it
running the engine on my laptop and see if that produces the error. On my
laptop, I get the warning that there are more parameters passed to the MH
command than are supported, but not the errors that you're getting.

I'm on the latest Master branch. Which one are you on?

-Gordon

Alex-Canales, 2016-08-02:
I am also on the latest Master branch and I have this issue. I run the engine on my laptop.

gordonab, 2016-08-03:
Ah, I see the error when using the previewer and understand the issue with
the get function. I'm rewriting it today to avoid this situation.

-Gordon

gordonab, 2016-08-08:
8976746d47b644324ea0fbe5a1612d5d8ceaa83a fixes this issue by using the current command position for the Z height instead of polling G2 for an update.

closed 2016-08-08 by gordonab