FabMo/FabMo-Engine #498: Up next roulette

SW0rden, 2016-08-09:
When dragging submitted jobs into the up next area, you get a loop going.  See attached screen capture.  You can also see the need to click a second time after deleting the one job and trying to drag the remaining one into up next.

Also, didn't the order that jobs are submitted in used to be that the first one would stay in up next and the others in order in the cue?  Now, as you submit them, the last one submitted goes into up next.  If you wanted them as they were submitted, you would need to reverse your order of submission or drag/rearrange after submitting all of them.

[FabMo_Roulette.zip](https://github.com/FabMo/FabMo-Engine/files/409250/FabMo_Roulette.zip)

ryansturmer, 2016-08-10:
Yes, the submitted order has changed.  We are in discussion about what the "correct" order should be.  What are your thoughts?

SW0rden, 2016-08-10:
I guess that if I'm submitting more than one job at the same time that are intended for running on the same project, my tendency would be to submit the one that I want cut first as my first submit and the one that I'd be cutting last as the last submit.  Sure, I could always drag and drop later if I didn't submit in the intended order, but that would be my normal/intended work flow.

ryansturmer, 2016-08-30:
I agree with @SW0rden in that new jobs should come in at the back of the queue (FIFO) - @wholetthedogsout can we get this changed for 1.2.1?

closed 2016-10-19 by ryansturmer