FabMo/FabMo-Engine #502: Unit Display in Previewer does not seem consistent

tedh-shopbot, 2016-08-12:
With FabMo set to mm, distances in metric file were shown in inches ... clicking the mm icon at the bottom changes the display distance to mm, but this seems out of sync with the default settings.

I think I would prefer the setting icon/toggles in previewer to display current condition, with a click on them toggling them to the alternate. I can see how there could be two options here, but in this particular display, I think it would be less confusing ???

Alex-Canales, 2016-08-12:
The viewer was designed to be the more possible independant from the FabMo environment. When I developped it, I thought it could be a could application to integrate somewhere in a Shopbot website for people not using (yet) FabMo and wanting a G-Code viewer. Therefore, it does not check any configuration made in the FabMo environment neither access to anything in the FabMo.

However, I can easily make sure the unit is automatically set to the user configuration.

ryansturmer, 2016-08-15:
I believe the current units come in the status report, so the report can be consulted to determine what the display units are currently.

Alex-Canales, 2016-08-15:
I am actually working on this issue, it should have been solved earlier today but I went into a bug.

For the moment, the unit displayed is corresponding to the unit set in the G-Code (or the converted SBP) file. By default, it is in inches.

I was going to use the units into configuration, under "machine" units, but I will use instead the units in the status report.

commit eec56526 (2016-08-16): Update viewer. Display units according to user configuration. Fixes #502.
closed 2016-08-16 by Alex-Canales