FabMo/FabMo-Engine #503: Making "Authorize" and the Green START Button more usable ...

tedh-shopbot, 2016-08-15:
We want people to use the Authorize system and not just disable it.

In that spirit, I think we want to try and eliminate some of the distractions. Many of these are specific to SBP files and programming things, But there are others that apply to the general case. For that reason, there need to be options for messages sent to G2 not to universally require authorization ... we need some sort of "ignore" flag because there are a lot  of distracting cases.

**Keypad Zeroing** should not require authorization
**Keypad Moves** there should be an option to turn off authorize for a number of relatively safe "convenience" operations, Keypad moves might be one ... maybe this could be _on a per session basis where 3 brief presses on the green button would release this subset  of authorizations_. The 3 presses would essentially be saying, "I'm here for this session with the tool and I take responsibility". There could be an indicator in the dashboard display (and in the LED's to indicate that Authorize is partially overridden).

For me, **using Sb4** would fall into this class. I would like to be able to send commands from the command line of Sb4 without having to hit the green button for every single one. I could make the time delay longer. But if I had something like the 3-press option, then I would be all set and not need to generally defeat the Authorize system.

Within and SBP file, once the file has started, you should not need to hit the green button as is currently required with the use of the PAUSE command. Just responding to the command should be enough.

For Apps that move the tool but do not turn on a spindle or router, the 3-press system should also disable the need to authorize for the session so that all the little house keeping kinds of things that need to be done can be done easily. Touch-and-go would be an example where after 3-press, it should just work ... not to mention various homing and zeroing routines.

**The one case that always requires Authorization** would be when a spindle is turned on in a file, thus we need to do a quick scan of the file. We could also designate a flag for file or App inclusion that similarly requires Authorization.

We definitely want people to be safe and use the system ... thus we need to work to eliminate the frustrations.

ryansturmer, 2016-08-15:
To clarify a little how things work, and continue the discussion i'll put some thoughts here with some additional questions.

First, to be totally clear - authorization is not required for all messages sent to G2.  Authorization is required in two instances (the distinction is important): 

1. To **execute a runtime command**: You are required to authorize the tool whenever you bring the machine from the idle runtime state to an active runtime to run a command. This covers the manual keypad situation, as well as SB4. (SBP and Manual mode are both runtimes)  Authorization is conferred when the button is pressed, and is revoked after the expiration time, which allows for the user to continue to engage with the tool for a time, but requires them to reauthorize if enough time has elapsed that they could have wandered away from their tool.
2. To **run a job** - you have to authorize the tool any time you run a job, no matter what.  Even if you've still got authorization time left after a runtime authorization.

Because zeroing is a runtime command (the direct execution of SBP code) it requires authorization.  A better strategy would be to allow the runtimes to specify whether or not a given command requires authorization.  This way runtimes that change the tool configuration or conduct activities that don't move the tool can do so without requiring a green button press.

To address specific topics:

**Re: 3-Button Press**  - Two questions:  (1) What should count as a "session" ?  What begins a session, and what ends it? (2) Allowing the defeat of the authorize system to run *any* SB4 command means you can even do big fast/dangerous moves that are not really in the category of relatively safe convienence operations.  Are we really ok with that?

The 3-Button press feature was actually implemented previously, in the form of the "pre-emptive" authorization - any time the green button was pressed, this was regarded as an authorization, whether the tool was "armed" with an authorization request or not.  This way, you could walk up to your tool, tap the green button, and begin work, without ever having to see an authorization dialog.  I see this as very similar to the 3-button press, and would be easy to implement again, but a few considerations:

1. On a large tool, the buttons are often large, bulky buttons that are a pain to press 3 times in rapid succession.  What about a button in the dashboard or DRO, that lets you request the temporary auth, and then a single press on the green button to grant it.  Or maybe that auth is requested when a user logs in?
2. The more we discuss this, the more I like the idea of communicating an "authorization key" that is ephemeral, and can be expired based on certain conditions.  For instance, if touch and go wants authorization for the session - it can request a session key.  If SB4 wants auth for a single potentially dangerous move, it can request that. The nice thing about auth keys is that you can tie them to specific devices, and expire them based on conditions relating to connections with those devices.  If I connect, open touch and go, and touch, it can request authorization - I hit the button, and then touch-and-go is unconditionally authorized, only from my device, until my device disconnects its session or i logout.  You simultaneously enhance security but make things more convenient and less of an encumberance. 
3. Remember that any "exceptions" to the authorization model that we give to SB4, or touch and go, or whomever, we are giving to every app ever.  We want to make it easy for users to be safe, but we also want to make it difficult for developers to be dangerous.

** PAUSE Auth Request ** We have the pause request require an authorization on purpose, because it is possible (and likely, especially on big tools) that you've wandered away from your tool when a pause is encountered, and you are possibly remote when the PAUSE is encountered.  It was seen as unwise for anybody to be able to press resume and set the tool into motion, when it's possible that nobody is physically coincident with the tool.  A few considerations there:

1. We could simply interpret the green button as "Resume" in those pause situations, taking out an extra click
2. We concieved of this saftey feature before we had user authentication.  Currently, only the logged in user can resume the tool, but we maybe should make it so that only the user that started the job can do that?  In that case, I'd feel better about lifting the pause/resume authorization requirement.
3. I think it makes sense to do **both** 1 and 2 above.  (Although 2 merits some discussion.  @jlucidar what do you think?)

## Executive Summary

Lots of rambling, but to sum up.

1. We should allow runtimes to distinguish between commands that require authorization and commands that don't.  That'll take some of the auth noise out of these commands that don't really have safety implications in their execution.
2. We should consider lifting the authorization requirement for pause/resume, but only in exchange for enhancing the security surrounding who is allowed to pause/resume the tool. (Make it so only the person who started the job or admin can resume it)
3. We should switch to a model that uses authorization keys, and have those keys contain rich data about what devices are authorized, and for how long.  Authorization requests can be more specific, and provide an authorization type that is more convenient (And doesn't encumber the user with incessant auth requests)
4. More discussion required on what we mean for the spindle-in-a-file authorization.

tedh-shopbot, 2016-08-17:
OK ... this is good and got me thinking about it. My responses are to numbers in the Summary.

First, let's keep 1-4 on the table while not necessarily making it more complicated just for the sake of it ...

#1 yes, we should make this one  so. It will address 80% of issues.
#1a. Then per your comments above, I think re-instituting the 3-Press Button press (it could be just 2, or something else ... just to make sure it is more than a tap and the user is aware they are doing something significant), which along with me making my own timeout longer ... would probably take care of another 15% of my issues. Which is probably enough!
#2. OK ... might be OK to allow given authenticate.
#3, #4 I could  see these as the planning going forward, yes ...

closed 2017-05-10 by ryansturmer