FabMo/FabMo-Engine #506: Configuration not updating correctly on unit change ...

tedh-shopbot, 2016-08-15:
V1.2.0 (and having just run handibot-production-8_1_16.fmu_w/restart)

Changing from inch to mm in unit default
  - if you immediately explore config app, you will note that "inch" goes to "mm" for chordal tol, but value does not change
  - no values change speeds or units
  - but if you close the app and then re-open it; values have correctly updated

Now, if ** instead of opening app again ** you turn off tool, and then restart it, default units is now inch but values are all appropriate for metric. At this point, you really go downhill if you try and change anything back to correct units.

SW0rden, 2016-08-15:
I've been struggling with this for a couple of weeks now as well.  Since the only way I can get my machine to stop the "short stop" jerk/ramp issue is to dump my config, reboot and fill in my values again, it has become rather difficult to enter values and have them updated properly.  See issue #488.

tedh-shopbot, 2016-08-15:
Yes ... it is almost as if all the dependent configuration changes do not get updated before you shut-down or reboot, then they all get totally out of sync with each other and it is hard to ever get it straight.

closed 2016-08-16 by ryansturmer

SW0rden, 2016-08-16:
Hmm, not fixed for me.  Go to config (after deleting G2.json) and fill in values, go to job manager and it hangs at the spinning gear.  Go back to config and it's set to mm unit system in general and all values are empty.   Try going back to job manager again and if it loads the manager this time, go back to config and the fields will populate after few seconds.    

v1.2.0-22-g5d68ca6

ryansturmer, 2016-08-16:
@sworden please furnish your logs, plus anything that is printed in the javascript console.  It looks like you're running a strange version - how is it that you're doing your updates?  Can you go to the updater and pull the latest developer version, and provide the log printout from that process as well?

SW0rden, 2016-08-16:
Attached are the console outputs after the last update approximately 3 hours ago and after the dev update.

My home/fabmo/log folder is empty, possibly because it's set to informational.  What should I set it to?  I see no other log folder/files.

I generally always update through the update console by choosing release candidate.  

[Last Update Ouput.txt](https://github.com/FabMo/FabMo-Engine/files/420993/Last.Update.Ouput.txt)
[Latest Dev Update Output.txt](https://github.com/FabMo/FabMo-Engine/files/420994/Latest.Dev.Update.Output.txt)

ryansturmer, 2016-08-16:
Try now that you're on the dev update.  You have not been getting the benefit of any of our recent changes.  We don't have a candidate for release yet.

ryansturmer, 2016-08-16:
Also, go ahead and update your firmware now that you've got the latest master, I'm not sure your old RC version has the latest firmware.

ryansturmer, 2016-08-16:
And for logs, you can just go to /log in the browser, to get the latest live log

SW0rden, 2016-08-16:
Ok, thanks.  Version is now v1.2.0-41-ged68f09.  Here's that log also.

[log.txt](https://github.com/FabMo/FabMo-Engine/files/421041/log.txt)

commit ed68f090 (2016-08-16): Fix #506