FabMo/FabMo-Engine #508: Lock-ups after Spindle-ON, "Waiting for FabMo"

tedh-shopbot, 2016-08-17:
I've noticed is that one is likely to get that same sort of lock-up that others have been describing, right after a spindle on command:  SO,1,1. I've been planning to try and figure out if it comes from the relay coming on or from the router powering.

*Just wanted to get this down somewhere. It has happened relatively frequently over the last 6 months.

ryansturmer, 2016-12-08:
Are we still encountering this issue?

tedh-shopbot, 2016-12-08:
I have not noticed recently. It was occurring frequently on the tool that did the same thing in the middle of a zeroing routine run -- that latter was caused by the USB cable wiggling.

I have had a lingering concern that doing something which produces a current spike, such as turning on the router, has been a source of G2 disconnects ... but have not current data and have not seen it recently.

closed 2016-12-08 by ryansturmer