FabMo/FabMo-Engine #518: OpenSBP error on commands w/o parameters ...

tedh-shopbot, 2016-09-09:
When a SB Command such as SA or ZT is issued with a comma it seems to generate an error. Don't know the impact of the error, and probably a bit of a pain, but "SA, " (with comma) and "ZT, " (with comma) should be acceptable and not generate an error unless someone tries to enter a parameter.

ryansturmer, 2016-09-09:
Can you describe the error?

tedh-shopbot, 2016-09-09:
debug: fsync()ed /opt/fabmo/config/instance.json [config]
warn: More parameters passed into ZT (1) than are supported by the command. (0) [sbp]
debug: Saving config to /opt/fabmo/config/g2.json [config]

ryansturmer, 2017-01-20:
This is actually a warning, not an error - I'm going to mark this as not critical for MAX release.

closed 2018-12-06 by tedh-shopbot