FabMo/FabMo-Engine #522: Unable to change unit values

SW0rden, 2016-09-13:
Version 1.2.2  OS 1.5.3  Logged in as admin with privileges.

I'm unable to change unit values and anytime you come back to the config page, the values are blank.  This is on a secondary unit that **does not** have the resistor to allow firmware updates removed.  Is that the issue?  Other than that, there's no difference between my two units.

Also noticed at the same time that on a PC and mobile in FF, the 'Update FabMo Software' button is grayed out, but in Chrome and Vivaldi it's not.

ryansturmer, 2016-09-15:
You need the latest firmware scott to get unit values, so that's more than likely your issue.  I'll open a separate issue for the update software button.

closed 2016-09-15 by ryansturmer

SW0rden, 2016-09-15:
Thanks Ryan, I kind of figured that might be it since this unit hasn't had that modification.  Although, this is the first I've noticed it and have previously been able to set the unit values.  Was this a recent change?

ryansturmer, 2016-09-15:
The new firmware was introduced to the build a few months ago.