FabMo/FabMo-Engine #528: App Routines in Fire-Up! don't run if Authorize is enabled.

tedh-shopbot, 2016-10-05:
Version 1.2.2
Don't know why no one has noticed this. 
If Authorize is enabled the green buttons such as Park or Center in the Fire-Up! App don't work.
Possibly because first lines are variables and not moves? Or, just because OpenSBP from within App?

ryansturmer, 2016-10-05:
Can you provide steps to reproduce?  Works on my 1.2.2 machine, except the first time I get the "Waiting for fabmo" which is known (and fixed in the rc) - I assume this is a different symptom?

tedh-shopbot, 2016-10-05:
Yep ... thought it was different ... but ...
On brand new tool and build 1.2.2
- start tool
-log in AP
-xy zero
-move around with keypad
-"Park" from Fire-Up! ... requests authorize then just goes back to app with no move. Same for home, etc ...
- Was thinking it might be related to those variables not having been initialized before macro run from app?? because I can't replicate on older tool at Handiland
- But this one behaving erratically after ... lot's of random log-in request and "waiting for FabMo" breaks
** I never notice some of this because I always have "authorize" off because it is so distracting and does not work on simulator.

ryansturmer, 2016-10-05:
I think when we coalesce the master with rc and release, this will go away, which should be soon.  @wholetthedogsout has fixed the logout issues (hopefully) and the variable initialization is fixed in the rc

closed 2016-12-08 by ryansturmer