FabMo/FabMo-Engine #531: preview displays extra line

jw4rd, 2016-10-18:
The extra line is not shown when the preview animation is active.

g20
g0z0.25
g0x0y0
g1z-0.125f30
g1x1y0f30
g0z0.25
m30

Alex-Canales, 2016-10-19:
Which extra line are you talking about?

jw4rd, 2016-10-19:
the line from x0z0 to x1z-0.125

Alex-Canales, 2016-10-19:
If you are talking about the extra line that should be displayed where I circled, this is normal. The line for the G-Code and the line for the Z axis overlap, therefore only the line for the Z axis is displayed.
[image]

jw4rd, 2016-10-19:
[image]

Alex-Canales, 2016-10-19:
OK, I was using the stand alone viewer app for testing your code. The viewer app in FabMo Engine is maybe not the more recent version. I will update and test it as soon as possible.

commit 86062129 (2017-03-20): Update viewer. Fix #531.

- Three.js and GCodeToGeometry are now added as node modules.
- Foundation dependency is removed.
- Removed very old version of G-Code Viewer.
closed 2017-03-20 by Alex-Canales