FabMo/FabMo-Engine #533: "Connect" button not connecting in Network Manager in 1.2.5 ...

tedh-shopbot, 2016-10-28:
I am now seeing this (@handiland) on several devices. FabMo in AP mode, go to Network Manager, select network and fill in passphrase, THEN no effect when hitting "Connect" button. I am able to get on network via updater.

Browser console something like ...
 '' because the form's frame is sandboxed and the 'allow-forms' permission is not set"

ryansturmer, 2016-10-29:
What browser/OS?

tedh-shopbot, 2016-10-29:
See additional discussion in FabMo Glip ...
All browsers, multiple computer, multiple networks ... Bill and Me ...
The weird thing, is that this was not happening with the same flash 1.7.0 (1.2.2) last week. I even went back to 1.2.0.
Thought my computer was somehow corrupted (browser full of the wrong cookies, or something), but tried to virgin computer, including a RPi that did the same thing in linux/firefox. All of my failures have been from my image install. But tested BrianBot yesterday (your install) and got the same thing (in that case, on my computer).

tedh-shopbot, 2016-10-29:
My procedure for installing image is to do the image, then put unit on network, then update engine. I did at least a dozen FabMos that way. Now same process, the network manager fails at the "Connect" button with the sandbox error (and pretty similar error in all 3 browsers). And, no problem joining network from updater. Updating to 1.2.4 or 1.2.5 does not alter problem.

ryansturmer, 2016-11-01:
This is resolved in 1.2.6

closed 2016-11-01 by ryansturmer