FabMo/FabMo-Engine #535: Need a method to utilize Keypad within an OpenSBP file.

tedh-shopbot, 2016-11-11:
Best to be able to use it within something like a pause, or a special pause. Typically needed to re-position for an offset or zeroing type of operation. 

Will be useful for Manual Tool Changes as well.

ryansturmer, 2016-12-08:
In the shopbot "classic" environment - if I encounter an SK in a file, and then use the keypad to drive the tool around, when I dismiss the keypad to resume the file, what happens exactly?  Do I immediately execute the next move in the sequence, or does the sb runtime drive the tool back to the position it was in before the SK was encountered?

tedh-shopbot, 2016-12-12:
You immediately execute the next _instruction_ in the sequence. It is assumed that whoever coded the file will get you back if that is needed.

ryansturmer, 2018-02-06:
Closed in the latest dev.

closed 2018-02-06 by ryansturmer