FabMo/FabMo-Engine #537: Files sent to server via FabMo should be gzip compressed on the client before sending

ryansturmer, 2016-11-16:
Client files should be communicated to the server as gzip and decompressed there, at least for job files.  This will improve performance for large jobs, since text compresses so well

jlucidar, 2016-12-20:
I will experiment with that, but it doesn't seems to be a common practice, and it will introduce a compressing delay on client side before sending the file (can be several seconds...). 
Gzip does not seems to be well implemented in javascript. I will try with [this package](https://github.com/imaya/zlib.js/blob/master/README.en.md) and report.

jlucidar, 2016-12-20:
no success with the listed library... I may try with an other library called [pako](https://github.com/nodeca/pako) (not gzip though...) tomorrow...

ryansturmer, 2016-12-20:
We'll have to evaluate the relative benefits of gzipping vs. taking the network latency.  If gzipping is so expensive that it is comparable or even makes the situation worse, then it's probably not worth doing.  The thinking was that since it's sparse text, it should gzip extremely well, and thus provide a savings.   Additionally, I think that gzip (or other compression) Should be an *option* - not the only way to send files (API support for post for other languages should not be encumbered with a compression step if they don't want to use it)

jlucidar, 2017-01-06:
Files over 2Mb are now compressed.

closed 2017-01-06 by jlucidar