FabMo/FabMo-Engine #555: Keyboard PageUP PageDN for Z move; PageDN triggers a screen scroll too.

tedh-shopbot, 2016-12-12:


SW0rden, 2016-12-12:
Mentioned this months ago and was going to mention it again as well after seeing it again lately.  Possibly related to this, on mobile, if you slightly slide your finger when on the up/down arrows, the page will also scroll.  I've had it happen a number of times on mobile where the Z will continue to plunge and drive into the work piece or table.

This only happens on the Home screen.

ryansturmer, 2016-12-12:
Only on the home screen, or in apps too?

SW0rden, 2016-12-12:
Only on the home screen for me.  I've tried it in other screens and individual apps that have scrollable content and those places don't scroll up/down with the pg up/pg dn keyboard combo or on mobile when slipping you finger up/down a bit while arrowing.

wholetthedogsout, 2017-01-03:
This should be fixed in the latest dev with the remake of the home screen.

closed 2017-01-03 by wholetthedogsout