FabMo/FabMo-Engine #560: progress ring missing in Edge

wlyoung, 2016-12-14:
When a file is running, the progress ring around the red square is missing in Edge. Newest development  version

jlucidar, 2017-01-12:
@wlyoung , I do have the progress ring around the stop button on my Edge browser. can you please flush all the cache and retry ?

wlyoung, 2017-01-14:
Hey Jimmy,

I was at ShopBot when I saw that problem and it was on the first FabMo
version that had Edge support. I'll check again tomorrow when I get back to
the shop.

Bill

wlyoung, 2017-01-15:
> mobile 757-999-5592 <(757)%20999-5592>

Hi Jimmy,

I tried it again this morning on a Windows 10 tablet and the progress ring
around the red square is still just grey...no indication of progress while
a job runs. I'm running the current release of FabMo though, the 1.8.0 usb
update and not the dev version, so it may have been fixed in newer versions?

Bill

jlucidar, 2017-01-16:
Yes, fixes for MS Edge have not been released yet, it is still on the dev branch. I will close this issue as it seems to work here. please reopen it if you encounter any problem .

closed 2017-01-16 by jlucidar