FabMo/FabMo-Engine #571: Recent file view in Job Manager; show last cut time

tedh-shopbot, 2016-12-31:
Also, for a file that is being replayed, show previous cut time in display. This let's people know how long file will be if specific knowledge available.

ryansturmer, 2017-01-03:
Do you mean the last time the file was run? Or the duration of the file the last time it was run?  I could see the "last time the file was run" being pretty useful for recent jobs, with "estimated duration" (the estimate coming from last time run) being useful for when the file is running.

tedh-shopbot, 2017-01-03:
Not sure I understand the subtlety of the question ...

ryansturmer, 2017-01-03:
It's a duration vs absolute time:

Last cut time: "4:00 Yesterday"
Last cut duration: "3:56:01"

tedh-shopbot, 2017-01-03:
Ahh ... well, the history gives you both pieces of info. Once you know when you last ran it, I don't think that is very relevant. But once you start re-running the file, I think it would be helpful to have the reminder of what the duration of the file was the last time -- so that is what I would display somewhere while it is running. And, as you note, on the recent file card, it might be nice to show when it was last run. Actually, one might as well put the duration there too. I am often frustrated by having to dig back into history to find out what the runtime of the file that just shut down was.

closed 2018-12-06 by tedh-shopbot