FabMo/FabMo-Engine #574: Any chance of improvements to startup speed (from point of re-powering)

tedh-shopbot, 2017-01-11:
This is one of the things that makes the system seem pretty unresponsive

ryansturmer, 2017-01-12:
Substantial gains in this area are probably going to be unlikely, but something that will probably help is the use of the LEDs to indicate the system state, which will be coming in the next major version.  On handibot, the yellow chasing LED will be the default on startup, and it will turn green after a completed boot.

tedh-shopbot, 2017-01-12:
How about something like firing up the dashboard before installing all the apps?

ryansturmer, 2017-01-12:
The apps are only installed the very first time you run the dashboard (at the factory) or when installed apps have changed.  Rob has suggested some kernel stuff that may improve the overall boot time, I will look into that.

closed 2018-12-06 by tedh-shopbot