FabMo/FabMo-Engine #577: Problem with Smoothsketch in Edge browser

wlyoung, 2017-01-15:
Using a Windows 10 touchscreen tablet, if I open Smoothsketch and try to draw something, the whole dashboard frame moves around within the window instead of drawing.

Windows 10 with latest updates, and current release version of FabMo from 1.8.0 usb install

jlucidar, 2017-01-18:
I'm moving this issue to https://github.com/FabMo/fabmo-smoothsketch-app/issues/ as it is app specific

jlucidar, 2017-01-20:
this one is giving me a hard time. I tested it on a surface pro, and it works fine with the stylus, but not with my finger. I wonder if this is not a feature that is surface specific, but I don't have an other windows tablet to try it on...
On my touchscreen, fingers are used to do navigation in Edge.

closed 2019-10-15 by tedh-shopbot