FabMo/FabMo-Engine #579: Slight wish ... last path memory for Android

tedh-shopbot, 2017-01-16:
When using the Job Manager with windows, there is good memory of the folder you are working in and after you browse to your first file, the workflow is very smooth. For Android, there seems to be no memory of the location of the last file called. It can be a pain if you are hidden multiple folders deep. Android seems to revert to some other apps last folder.

jlucidar, 2017-01-18:
This is not something we have control over. It is browser dependant, and we don't have any way to specify which folder we want to open...
There are some detailed explanations of the problem here : http://stackoverflow.com/questions/20537696/remember-and-repopulate-file-input

closed 2017-01-18 by jlucidar

tedh-shopbot, 2017-01-18:
The response linked above is interesting. It argues that you can't do it because it would be insecure and a huge violation of internet robustness. That does not explain why Windows does do it so well. I'm thinking it is because the question is misunderstood. What must be happening is not that the calling agent (browser) is requesting a defined path or file (deemed dangerous), rather that on the previous call, it set a flag/cookie/whatever that causes local device to start using the most recent path. This must be the case, because the path it is using on my android is one that was set from a "browse" call from another, non-FabMo, app ...  It does not start from root.

jlucidar, 2017-01-18:
yes, it is a browser intern feature but it's not configurable. I don't know what logic is behind chrome on android. The thing is that there is not a single default file manager on Android, so my guess is that it is also depends on the brand of your phone.