FabMo/FabMo-Engine #581: Bit random ... password hints in Network Manager

tedh-shopbot, 2017-01-18:
A few months back, when logging into a network via that network manager, I used to get a list password hints/fill-ins that typically contained what I needed. Now I get a list of hidden "passwords for user:" that appear to be the passwords I used to log into the tool, not the password needed for the network (the network does not have users). I don't know where this stuff is or if we can manage it ... but it is sort of distracting. Maybe we can turn off if not able to revert to previous behavior. Happens in chrome on both PC and Android.

jlucidar, 2017-01-18:
This is something handled by chrome and we don't have control over autocompletion. I will force it off.
http://stackoverflow.com/questions/2530/how-do-you-disable-browser-autocomplete-on-web-form-field-input-tag

closed 2017-01-18 by jlucidar

tedh-shopbot, 2017-01-18:
It is interesting that they changed how it works.

jlucidar, 2017-01-18:
well the autocompletion in the network manager is actually conflicting with the one from the user login. It worked before because there was no user management...

tedh-shopbot, 2017-01-18:
ahhh

commit 43cc0dd7 (2017-01-18): fix #581