FabMo/FabMo-Engine #593: If not currently within an App, would it be useful for any refresh to take you to your start-up App?

tedh-shopbot, 2017-03-14:


ryansturmer, 2017-05-12:
There's no such thing as not in an app, at this point.  It's an important discussion, though.  When should a refresh keep you in place, and when should it take you somewhere else.  For system apps, we strive for a refresh to leave you where you were, which means remembering your state at all times, which can be burdensome.  There might be some state memory stuff we could do in the dashboard scaffolding, and perhaps hooks we could put in the apps that indicate "yes I am the type of app that will remember my state and can suffer a refresh" or "no I am not if the user refreshes please go back to their default app" - does that make sense?  The use case is, for instance if you refresh by accident - ideally you wouldn't lose your place in the application.  This is how most web applicatons are designed to work.  I am going to chance it by reloading this page before I submit this comment.

.. yeah - see my state was retained.  I'm right where I was.  I didn't go back to the github homepage - and even my text was retained, which I do not think is a function of my browser - there was a slight delay before it appeared.  The page is keeping it, willfully, and putting it back in place so I can continue to work on it.

In our case, I think there are arguments for sometimes you want to go home, and sometimes not.

tedh-shopbot, 2017-05-13:
I think I jotted this down too fast. As noted, I think a "refresh" should generally refresh where you are.

What I meant (I don't know why people don't understand me) ... was that I think that some people might want to designate an alternate "home" page as well as an alternate "start" page. For example, on a tool like that chop saw out back, after finding oneself in another app, such as the job manager, you might want to easily and always return to your standard interface app (chop saw) rather than having to go to the "apps app -- current home app" and re-select your default app every time. This behavior is what makes using the (abbreviated) Fire-Up start page awkward to use because you always have to go home then to the app to get to it. 

What I thinking I want to try is being able to set a start and a home app. When you set a home app that is not the apps-app, we could just add an apps icon to the left panel.

closed 2018-12-06 by tedh-shopbot