FabMo/FabMo-Engine #599: Would be nice if "collapsed"/"not-collapsed" left menu state were remembered across sessions per device ...

tedh-shopbot, 2017-03-31:


tedh-shopbot, 2017-04-01:
Actually, just noticed that even a refresh wipes collapse state.

For me, primarily working on tablet at handiland and PC at home, once you know the icons, the screen looks quite attractive in the collapsed state. I'm very comfortable working on it and don't feel all scrunched- in. Thus I like the idea that this is my own default. And, it's a pain to have to keep setting it.

Also, in the best of all worlds ... I'd have a simple tool-tip in the collapsed state, just in case a reminder is needed, and a more expansive tool-tip in the open state to let people know what's in the feature. For me, after thoughtful layouts and expected functionality, helpful hints are the next most useful self-documenting feature of the interface.

ryansturmer, 2017-04-14:
How would you access the tooltips on mobile?

ryansturmer, 2017-08-10:
The proposed fix after talking with @wholetthedogsout is to keep the mobile behavior as it is, but for tablet sized displays and up remember the state of the collapsedenessss of the menu.  Is this consistent with your expectations of how it should work?

(With further work to continue after the next release on hinting, tips, menu reorganization, etc)

closed 2018-12-06 by tedh-shopbot