FabMo/FabMo-Engine #607: Crash when moving manually

Alex-Canales, 2017-04-30:
In Configuration->Machine->Authorization Button, "Authorization Required" is unchecked.

When moving manually, using the keyboard or the buttons, while I go to a single and same direction: it works. If I move to another direction, nothing is done. Here is the complete log:

```
null
info: Arming the machine(No action) [machine]
info: Firing automatically since authorization is disabled. [machine]
info: Machine is authorized for the next 30 seconds. [machine]
info: Got a machine state change: idle [machine]
debug: Saving config to /opt/fabmo/config/instance.json [config]
debug: fsync()ed /opt/fabmo/config/instance.json [config]
info: Got a machine state change: idle [machine]
warn: Got a state change from a runtime that's not the current one. ([ManualRuntime]) [machine]
info: Got a machine state change: manual [machine]
Creating a cycle context.
info: Creating promise 4 [g2]
info: Priming driver (manually) [g2]
error: Not writing to gcode due to lapse in responses [g2]
error: Not writing to gcode due to lapse in responses [g2]
debug: Saving config to /opt/fabmo/config/instance.json [config]
error: 132,STAT_MCODE_COMMAND_UNSUPPORTED,M command is not supported [machine]
debug: fsync()ed /opt/fabmo/config/instance.json [config]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
warn: Not quitting because a quit is already pending. [g2]
warn: fixedMove(): Not moving, due to already moving. [manual]
warn: fixedMove(): Not moving, due to already moving. [manual]
info: Authorization timeout (30s) expired. [machine]
info: Machine is deauthorized. [machine]
null
error: Error: Cannot arm the machine from the manual state.
    at Machine.arm (/home/alex/Shopbot/Engine/machine.js:248:9)
    at Machine.executeRuntimeCode (/home/alex/Shopbot/Engine/machine.js:652:9)
    at Socket.<anonymous> (/home/alex/Shopbot/Engine/routes/websocket.js:143:13)
    at Socket.emit (events.js:95:17)
    at Socket.onevent (/home/alex/Shopbot/Engine/node_modules/socket.io/lib/socket.js:330:8)
    at Socket.onpacket (/home/alex/Shopbot/Engine/node_modules/socket.io/lib/socket.js:290:12)
    at Client.ondecoded (/home/alex/Shopbot/Engine/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/home/alex/Shopbot/Engine/node_modules/socket.io/node_modules/socket.io-parser/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/home/alex/Shopbot/Engine/node_modules/socket.io/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/home/alex/Shopbot/Engine/node_modules/socket.io/lib/client.js:175:18) [websocket]

```

Below are the explanations of the log. 

This part is when I press the button for going to the first direction then I release the button:

```
null
info: Arming the machine(No action) [machine]
info: Firing automatically since authorization is disabled. [machine]
info: Machine is authorized for the next 30 seconds. [machine]
info: Got a machine state change: idle [machine]
debug: Saving config to /opt/fabmo/config/instance.json [config]
debug: fsync()ed /opt/fabmo/config/instance.json [config]
info: Got a machine state change: idle [machine]
warn: Got a state change from a runtime that's not the current one. ([ManualRuntime]) [machine]
info: Got a machine state change: manual [machine]
Creating a cycle context.
info: Creating promise 4 [g2]
info: Priming driver (manually) [g2]
error: Not writing to gcode due to lapse in responses [g2]
error: Not writing to gcode due to lapse in responses [g2]
debug: Saving config to /opt/fabmo/config/instance.json [config]
error: 132,STAT_MCODE_COMMAND_UNSUPPORTED,M command is not supported [machine]
debug: fsync()ed /opt/fabmo/config/instance.json [config]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
info: Priming driver (manually) [g2]
warn: Not quitting because a quit is already pending. [g2]
```

This part is when I try to move to another direction:

```
warn: fixedMove(): Not moving, due to already moving. [manual]
warn: fixedMove(): Not moving, due to already moving. [manual]
```

Then comes the authorization timeout:

```
info: Authorization timeout (30s) expired. [machine]
info: Machine is deauthorized. [machine]
```

And what happens after that each time I press a move button:

```
null
error: Error: Cannot arm the machine from the manual state.
    at Machine.arm (/home/alex/Shopbot/Engine/machine.js:248:9)
    at Machine.executeRuntimeCode (/home/alex/Shopbot/Engine/machine.js:652:9)
    at Socket.<anonymous> (/home/alex/Shopbot/Engine/routes/websocket.js:143:13)
    at Socket.emit (events.js:95:17)
    at Socket.onevent (/home/alex/Shopbot/Engine/node_modules/socket.io/lib/socket.js:330:8)
    at Socket.onpacket (/home/alex/Shopbot/Engine/node_modules/socket.io/lib/socket.js:290:12)
    at Client.ondecoded (/home/alex/Shopbot/Engine/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/home/alex/Shopbot/Engine/node_modules/socket.io/node_modules/socket.io-parser/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/home/alex/Shopbot/Engine/node_modules/socket.io/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/home/alex/Shopbot/Engine/node_modules/socket.io/lib/client.js:175:18) [websocket]
```

Tested only with the controller, not on real machines. The log displays nothing unusual.
Tested commit: 7097b349cd9c9a998065d8f11ca693587eabbb55

ryansturmer, 2017-05-08:
What version of the firmware are you running, Alex?

Alex-Canales, 2017-05-19:
No firmware is detected. It may be an hardware issue, it happened right after I traveled. I am closing this issue the time I reflash the controller and will reopen it if it happens again.

closed 2017-05-19 by Alex-Canales