FabMo/FabMo-Engine #619: Hitting stop in zeroing routine loses 0 ... (Don't know if there is anything that can be done)

tedh-shopbot, 2017-09-18:
If you are partway through Macro 3 (the Z axis will be only slight away from the hard stop) and you hit STOP. You will cause the Z to top out and lose location. Don't really see a way around it when we are using proxes or handibot-bump to zero Z at the top.

If we assume that Z TBC 0 is always at the top, then we could apply the pull-up height to the current Z offset ... which might be good protection for all. But it does presume that ZTBC = 0 ...

ryansturmer, 2018-09-20:
@tedh-shopbot is this still an issue?  I think the new probing plus our improved table coordinate system resolved this one awhile back?

tedh-shopbot, 2018-09-21:
Have no idea, but will give it a try.
Pretty standard for people to start zeroing and then realize they weren't quite ready and want to stop. So in Sb3 we sweated getting zeroing to stand up to a lot of interfering key-pounding.

closed 2018-12-06 by tedh-shopbot