FabMo/FabMo-Engine #621: More general handing of quoted statement in OpenSBP END instruction ...

tedh-shopbot, 2017-09-25:
END "_some text_"  ... is useful, particularly in having the simplified single response > quit
But using it comes with an error message.
How about skipping the error message, then it can be used more generally.

For example, providing:  "Zeroing Complete. Remember to REMOVE CLIP!"  ... > quit

For the error on end condition, we can simply do:  "ERROR! Input Switch is already tripped." ... > quit

[image]

tedh-shopbot, 2017-09-25:
The only thing we can do at the moment is end with a PAUSE "xxx" that then displays Quit/Resume, which is confusing at the end of an action.

ryansturmer, 2017-12-03:
What about END and FAIL?

`END "Thanks for playing"` - Ends the file, with a message, but not one that is labeled as an error. Job is considered "complete" in the history. (Ending message has no resume button)
`FAIL "Sorry you lost"` - Ends the file, with a message, but that message is presented as an error.  Job is considered "failed" in the history. (Fail message also has no resume button)

END would mark the end of a file, but no message.  Job is considered complete.
FAIL would mark the end of a file, but no message.  Job is considered failed.

gordonab, 2017-12-03:
Sounds like a great way to indicate the ending status of a file or macro.
Would this status then show in the job history?

ryansturmer, 2017-12-04:
The job history thing is already implemented.  If you `END` a file, it's marked as completed, but if you `END "Message"` it's marked as failed.  This update would give more options (basically giving all four options, to fail or not, and to message or not.

ryansturmer, 2018-09-12:
Just resolved this in another issue.

closed 2018-09-12 by ryansturmer