FabMo/FabMo-Engine #628: Easier log on in AP mode

tedh-shopbot, 2017-09-29:
When you go to AP on a device, you are required to enter logon to the wireless device ...
Then, immediately when you go tool, you have to enter the same pwd again for fabmo.

The double entry makes sense for joining networks where the credentials are different ... but it gets very old every time one goes to AP.

ryansturmer, 2017-12-03:
Those passwords are only the same because you have set them to be the same.  Remember that FabMo is actually a multi-user system, and the wireless key for AP is not the same thing as user credentials for logging in.

I agree, though, that the login prompt comes up too frequently - the login should be retained between sessions in AP mode, but because of the way the credentialing is handled, they're not, currently.  In an upcoming release this will be fixed.  (The "double" entry won't be removed, but the second entry will effectively go away as it will be remembered, even if you disconnect and reconnect)

tedh-shopbot, 2018-11-29:
no longer relevant ...

closed 2018-11-29 by tedh-shopbot