FabMo/FabMo-Engine #631: Need ability to disable "authorize" in apps, files, macros ...

tedh-shopbot, 2017-10-06:
Any sort of interactive app, file, or macro ... becomes virtually impossible to use with "authorize". You should be able to turn this check system on and off in apps, files, and macros. And, at the end of a file (or broken file, etc.), there should be an auto-reset to the state of "authorize" config that was present at the start of a file.

This feature will make interactive routines usable and will still protect the selected level of safety settings. Note that interactive routines, by definition, require the user to be at the tool to signal resumes and continuations. [Note that green button and resume on screen should be interchangeable.]

sjkelly, 2020-10-09:
Is there a status update on this? I have been able to get Fabmo running on a Pi4 and it appears to talk to the g2core device I am using. Though my setup doesn't have any interlock, so I cannot access the jog panel or send commands.

tedh-shopbot, 2020-10-09:
Hi Steve,
-Go to the configuration manager > the gear icon on the left
-Select machine > and scroll down to > Safety >> and unclick the authorize check box.

sjkelly, 2020-10-09:
Worked great! Thanks!

tedh-shopbot, 2022-02-15:
Putting in an updating comment and some tags. This issue is not about being able to turn "authorize" off, but to be able to have authorize ON when to approve the running of a file or app, but then optionally, programmatically  turned off within the file so that the use of queries of macros does not become unmanageable. Not quite sure how it would be implemented -- probably worth determining how problematic the situation is.

RobMackie, 2023-04-28:
imagine an AO (authorize override) 
AO MS, 5