FabMo/FabMo-Engine #637: Files pause and FabMo locks if input 4 is triggered during a file

davepreiss, 2017-11-14:
Currently have my proximity switches hooked up to input 4 if they are triggered while cutting the tool will pause (green resume and quit buttons) but FabMo will become unresponsive. Should be pretty easy to reproduce! Running rc-1.6.0.

ryansturmer, 2017-12-03:
By default, input 4 is configured as a stop input.  Can you confirm this is the case on your machine by going to configuration -> inputs?  The unresponsiveness is probably due to the fact that the system expects that to be a momentary button, and if you're hitting it with your prox, it's actually staying triggered - try changing the  number 4 input action from "Stop" to "None" and see if it improves your situation.

wholetthedogsout, 2018-03-23:
Tested. In above scenario machine will pause. Clicking resume or cancel will result in a cancel but fabmo will not hang. Closing for now may want to reopen if this is not the behavior we want.

closed 2018-03-23 by wholetthedogsout