FabMo/FabMo-Engine #639: Left side menu / DRO enhancements

ryansturmer, 2017-12-03:
I would like to propose some enhancements to the left side menu and DRO, that would represent a pretty substantial change in the flow of the UI (for the better, on both desktop and mobile)

The changes broadly, would be as follows:

1. Remove the DRO on the right side
3. Change the left side menu to be tabbed (with two tabs)
2. Take the position/status components of the DRO and put them in a "tab" of the left side menu
4. Make the main/default tab of the left side menu be more "action" focused. - *Run* job.  *Configure* machine.  *Edit* macros, etc.
5. In line with the action focus, put a job drop control in the left side menu, at the top. Should support D&D, multiple files, support upload progress indication, etc.
6. LSM should collapse completely with a hamburger.  No abbreviated sidebar - it's either there or its gone.
7. LSM will almost or totally cover the display on the smallest mobile devices.  This is expected, and the design of the rest of the UI should count on this being the case.

tedh-shopbot, 2017-12-03:
Would be interesting to see how this looks. I'm worried from the way it sounds that you might have to click twice for things that now take one click.

#616 was just about a much more mundane user-expectation issue with respect to remembering preference for short or long display.

ryansturmer, 2017-12-04:
This isn't meant to replace #616 - but would make it obsolete as the collapse behavior would be eliminated.

The idea here would actually to be to reduce the number of clicks for the most common workflows.  Loading and running files for instance can happen in the left side menu, eliminating the need to go to the job manager for that activity, which is what you do most of the time once you've got the tool configured.  The DRO function would be a single click (or two on mobile, with the hamburger) and would really be something you'd interact with pretty rarely - more common would be manual mode, which would be an "action" in the first menu, which would be single click to activate (again two on mobile if you count hamburger, but that's not any worse than now) - I haven't addressed the top-bar here, which could perhaps be used as a shortcut location for things like manual mode.

I will post some sketches here once I get them down and scanned.

ryansturmer, 2017-12-04:
[image]

Here's some mock-ups of proposed left side menu enhancements.  There's a drop target/button that lets you submit jobs automatically, that are queue-sensitive.  The "up next" item from the queue would appear if the queue is non-empty, and if there's more than one thing in the queue, a link to the JM for managing the queue is presented.  Menu items are more action focused, and there are fewer of them.  Management of macros and network move into the configuration app, and the idea is that the configuration app would become more legible and make it easy to get into and manage those facets of the app.

tedh-shopbot, 2017-12-04:
Looks promising.

ryansturmer, 2017-12-05:
For the DRO:
* Zeroing buttons for each axis
* Zeroing buttons for XY and XYZ
* Some mechanic for letting you enter a number for an axis and choose "set to this position" rather than "go to this position"
* Inputs/outputs displayed as now, but also optionally with names - ie you can go in the settings and choose a name for output 7, (say "Dust Collection") and you'd get a friendly placard to that effect in the DRO (in addition to the vanilla number row)
* Perhaps colorization of the position display to distinguish positive from negative numbers
* Perhaps a button to enact "manual drive" (although probably not - this button will probably be on the dash at all times, not just when DRO is visible)

SW0rden, 2017-12-07:
Ryan,

What side would the DRO be on when open?  A lefty would probably be ok with it on the left, but I feel it would be cumbersome for a righty.  For instance with a tablet in your left hand and reaching across to the the left for keypad usage.  Maybe a config setting for left or right DRO display.

Something else that I've though about in the past and your suggested changes to the DRO has brought it back to mind, would be more control over move/jog with the keypad.  Many gcode controllers/MPG's will have the option to set 'Jog' (which there is no jog option now) and also to set continuous or step.  Along with the step setting, there are generally three increments that can be chosen with a single to triple click and these could be set in config as well.  Something like 0.1, 0.01, 0.001.  This along with continuous jog allows for quick and precise positioning.  As it is now, it's hard to hold an arrow just the right amount to get close and then only have one increment choice (without having to go into config and change the value each time) with a single tap to quickly set it to the mark.  While this may not be an issue on the tiny envelope of a handibot, it soon becomes an issue on a larger tool.  If there are intentions of using FabMo on larger tools along with DT/DTmax, this would be very beneficial (I'd say a must).

ryansturmer, 2017-12-29:
@SW0rden thanks for weighing in.

So, the idea would be that the keypad would be moved OUT of the DRO, although I didn't mention this explicitly above, and would be reimagined as its own modal dialog that would essentially take up the entire screen on a mobile device.  THe newer, richer keypad would include more robust motion controls and support things like increments, feedrate scaling, stored positions, etc.  The modal would be a center-screen affair on a PC or large tablet, but on mobile, it would be large enough as to consume the screen, and would be equally accessible for lefties and righties.

SW0rden, 2018-04-16:
I haven't spent much time with FabMo of late, but finally took some time today to try out the dev version.

A couple of thoughts on the keypad updates...
For one who will most likely always prefer to use this with a keyboard and if you are looking to bring over some of your current SB3 users in the future, there needs to be a shortcut key like (K) and the ability to use arrow/pgdn pgup keys.  Unless I'm missing something, you can not use them now.  !?!

I feel that you should have a separate Z speed slider or I'd prefer a box to enter the values.  Sliders are fine, but I think a slider with text box to direct enter a value would be nice.  With the Z axis being a shorter stroke, it would be easy to move to far with the same speed as the XY. 

For PC use, it would be nice to see a decent larger DRO.  Since that part has been removed, all a PC user is left with is the small one in the upper right corner.  For someone with aging eyes, it useless at a glance.  Perhaps adding it to the running file screen would be one place.  I know that wouldn't be optimal for mobile, but if this ever makes it's way to the larger tools and as an update for current users, it's something that should be kept in mind.

wholetthedogsout, 2018-04-18:
Thanks for the feedback Scott. Much appreciated. 

The keyboard is coming back soon. And separate slider for the Z is in the works. It is challenging with the limited space on the phone (which the slider was mainly designed for) to fit another input box, but this can be easily incorporated at tablet size and up if you think it will be more useful there. 

As for the DRO it suffers from the same real estate limitations on a phone. No reason this can't be bigger on a desktop though. Will start working on a mockup of this.

Thanks again for the feedback.

SW0rden, 2018-04-19:
While using a mobile device is "cool" or "fun" for a bit, I much prefer to use a PC/laptop and going forward to where FabMo moves to the larger tools and in varied situations/businesses, I would think that sentiment would also apply.  I'd like to see the option to have both mobile and desktop versions, much like where you can choose to use the desktop or mobile site.  This would give users that prefer to use PC/laptop a full featured version instead of an abbreviated interface.

Instead of "Home" having seldom or just occasionally used apps, I'd like to see the Home page be what I would call the "Run" page with full controls, file loading, and DROs.  Take a look at the 2017 screenset for UCCNC http://www.thecncwoodworker.com/2017.html  I have been using this in tandem with FabMo on a 12" Lenovo Yoga touchscreen laptop and I keep going back to UCCNC.  Arrow/pgup/pgdn key movement is always available with the side slide out giving more granular (fixed) control when desired.

The "Apps" page could then have it's own page. 

Just some thoughts, I know that you guys have been working hard on FabMo and have had some setbacks, but I fear the interface is trending more mobile than admitted and once moving over to the larger tools (even DT/DTMax), will be an issue to us long timers/old timers that prefer a full, dare I say, more serious interface than the handibot crowd.

ryansturmer, 2018-06-28:
Scott these are all good points, and some alternative interfaces that put more "at your fingertips" are in the works.  The app we've developed internally for running some of our bigger tools is trending in that direction, and a more complete, full featured manual drive experience is sorely needed. 

Worry not about moble - mobile compatibility is essential, and we're always keeping our eye on it, but we have no intention of sacrificing "desktop" experience for it.

SW0rden, 2018-06-29:
Sounds good.

closed 2018-12-06 by tedh-shopbot