FabMo/FabMo-Engine #653: Keypad motion stutters with latest dev

SW0rden, 2018-04-16:
Keypad motion with v1.6.2-gb72b8cb-dev stutters and hangs when tapping arrow keys or clicking with mouse on PC.  Hangs and drives axis into stops at times.

Still waiting for keyboard shortcuts. ;-)

wholetthedogsout, 2018-04-18:
What device are you using when you are getting the hangs?

SW0rden, 2018-04-19:
With two different PC's, an Android tablet and phone.  Tapping an arrow key, it will move a little and stop and move a little more stop and may continue to move when releasing and then run into stops..

ryansturmer, 2018-09-12:
Calling this fixed in 1.7.0

SW0rden, 2018-09-20:
Basically the same as the defaults for a Desktop (Z is higher unit values due to having a different lead screw pitch at the moment).  It's not a stepper stall or lost steps, it's just a stop and go motion.  Tap and hold an axis button and it will start to move a bit then stop then start again then stop and so on for as long as you hold the button down or until you run out room and have to let up.

ryansturmer, 2018-09-24:
How are you connecting to the tool (wifi, ethernet?) and this is 1.7.0 i assume?

SW0rden, 2018-09-24:
WiFi and yes, 1.7.0.  It's the same with any device/PC on the network.  Touch and hold an arrow button or keyboard key and the motion is like __ __ __ __  Due to the shorter travel on the Z, if you are not quick enough to get it stopped, it will slam it into the table or top out.

closed 2019-10-15 by tedh-shopbot