FabMo/FabMo-Engine #654: v1.6.3 top bar/panel issue

SW0rden, 2018-04-20:
Top panel or whatever you call it is, well expanded or something on all pages (see attached).
Keypad is missing.
Double vertical scrollbars. 
?!?! :-( 

[image]

wholetthedogsout, 2018-04-23:
Looks like you have cached elements from the dev.  You need to refresh the url. Delete everything after the IP address in the address bar and hit enter.

SW0rden, 2018-04-23:
Nope, did that, same way on all devices and 3 browsers.

wholetthedogsout, 2018-04-23:
Your update must have failed then. You are definitely not getting all the files from the server. Try updating again. I just published a newer version of 1.6.3. So hopefully that will work.

SW0rden, 2018-04-23:
Works now, thanks.

closed 2018-05-02 by ryansturmer