FabMo/FabMo-Engine #657: History enhancement

SW0rden, 2018-05-14:
While looking through the file history and deleting some old ones, having to double click on the 3 dot menu to activate it, I was reminded of how easy it is to select multiple emails for deletion.  Could the history page be similar to the new Gmail and Outlook web mail where you could check mark several files to delete or if desiring to delete just one file (or perform other operations), as you hover over it, you would be presented with the options.

ryansturmer, 2018-06-28:
Why delete old ones?

wlyoung, 2018-06-28:
In my case I have a couple hundred sequentially named Badger files in the history that I know I won't need any more, with the occasional useful file interspersed that I need to run again. It would be nice to easily be able to delete big blocks of obsolete files, to make it easier to rerun the good ones

wlyoung, 2018-06-28:
I understand that's it's meant to be a history and not a file system, but if you're using a device like a phone it's tricky to reload a file if it wasn't saved locally

wlyoung, 2018-06-28:
Closed by accident...

ryansturmer, 2018-06-28:
Multi-select is an option we can pursue, I think improved organization of the history is important here too.   You should be able to select all the files that came from a single app, or were run before a certain date/time, or that fit other specific criteria.  It should also collapse repeat runs of the same file into one line item, since having it on multiple lines doesn't add any value - is there any reason to keep badger files in the history anyhow?  Should we give apps the option of setting an expiration date on their jobs so that they flush themselves out automatically without having to fall off the cliff natually with the storage space (which takes forever) ?

wlyoung, 2018-06-28:
Occasionally there's a need to recut one, but there is no reason to keep them after the next one is cut. I'd be happy as a clam if they disappeared automatically.

Also it would be nice if multiple runs of the same file didn't generate a new identical entry in history? One copy needs to exist for the next run I guess...could they all congeal somehow as one listing that was collapsed?

wlyoung, 2018-06-28:
(I missed your comment on repeated runs in one line. Perfect

SW0rden, 2018-06-29:
Why delete.... I run a lot of 3D files and they are generally 'one of's' and can be substantually larger in file size.  I don't know what the storage capacity is, but don't see any reason to keep them resident taking up space.

Multi-select for deletion would be a good.  With the case of often used files, it might be nice to have a 'Favorites' kind of folder/tab for quicker access as well. 

+1 on the duplicates.

wlyoung, 2018-06-30:
I'd like to be able to edit a file and have the option to completely replace the existing one so that they don't both exist in history. The same with uploading a new version of an existing job...I'd like to be able to replace the original.

It's too easy to forget to add a description when you're editing

wlyoung, 2018-06-30:
I'd also like a trash can to completely wipe out the history. I can imagine something like a teacher wanting to wipe the history at the end of the day?