FabMo/FabMo-Engine #661: nodejs version depdendency?

scotthz, 2018-06-30:
The [README](https://github.com/FabMo/FabMo-Engine/blob/master/README.md) says that the supported nodejs version is `v0.12.7`. 

Is that still the case? That is pretty old. I'm trying to get FabMo running on a Raspberry Pi with G2 on a Due, configured for my own hardware. Not having much luck at this point, but was trying so far with node `4.8.2`. Wondering if I need to back way up to `v0.12.7`.

Thanks in advance.

scotthz, 2018-07-02:
Thanks. FabMo seems to have a little recent activity, but it's been pretty hard to get started, at least coming from a non-Shopbot/Handibot perspective. Not much in the way of getting-started docs that I have found.

I did manage to get a working FabMo running on a Pi 3 with a fairly recent Raspbian release from April 2018 (just noticed that a new release dropped a few days ago). I got an example app installed and running after finding  and dealing with the fact that FabMo disregards the compiled-in G2 axis, motor, and input settings and unilaterally overwrites them when it starts up.

I'm running G2 version 0.99, build 101.03 with external drivers (I built the G2 image myself).

FabMo seems really appealing, but it's slow to get started with it. If I keep going in this direction, maybe I'll try to document my experience for others.

ryansturmer, 2018-07-12:
Hey @scotthz thanks for your interest.  I understand that it's slow to get started.  We're resource strapped and have been focused on internal development for our products, which we hope to be changing pretty soon.  In the mean time, let me see if I can provide some guidance.

The node version that we use is 0.12.7.  The reason for this is being locked into it on some of our older hardware.  We have a new hardware prototype coming in soon, that when we start to go into production with, we'll be doing a package audit and getting our node version up to modern standards.  It's not our priority at the moment, but should be happening soon.  We have run it against node version 4, which the last time I checked (a few months back) was what came stock on a raspberry pi 3 image.  It works fine with that, and if you find that it comes up and runs, I'd say go for it.  I know that there are issues on very new node versions, but I haven't gone to sort those out yet.

We do not have a raspberry pi image that is maintained and up to date, because we have no intention of using actual raspberry pi hardware in our products.  We have installed it here though, and I can confirm that it works.  The last time I did it, it was a matter of simply installing node 4, pulling the project dependencies with npm, and running the engine.  If you try that and get stuck, I'm happy to answer specific questions here.

We are going to work more on making the system more approachable - we are hoping to get some of the bugs ironed out and our focus more oriented toward improving the plug-and-play aspect of things.

ryansturmer, 2018-07-12:
And rest assured, there's lots of activity here.  Much of it is in our documentation and release preparation internally, and on apps, and other things that don't appear here in the tracker.

jlucidar, 2018-07-12:
Hey @ryansturmer, I can give it a look if you want. I've been playing with node.js 8.x for a while on my other projects. The transition should not be such a big thing, mostly I've been struggling mostly with broken Buffer apis. I will have some spare time over the summer, so I will probably play with FabMo a bit; It's been a while !

scotthz, 2018-07-19:
@ryansturmer ,
Thanks for the followup. Got sidetracked by my day job and just getting back to this now. I did get fabmo-engine running on a pi with node v4 and played some with the example app. Was pleased that it mostly "just worked" with my one-off G2 build, once I understood that fabmo wants to manage the fundamental G2 settings and overwrites things like motor, axis, and input configuration. I may be able to help towards making fabmo more approachable by the non-shopbot community, i.e. people like me. I've often been in similar situations where I need to come up to speed with something that wasn't specifically targeted at my use-case and have developed some coping mechanisms...

Is there a preferred forum for asking questions that don't assume that I'm coming from a shopbot/handibot paradigm? Do you have ideas or suggestions regarding how I might contribute as I move forward?

closed 2018-07-24 by scotthz

SketchThis, 2019-11-12:
I myself just went down this rabbit hole and came here to find the same answers.... 

Like @scotthz I'm trying, unsuccessfully to get the same setup workin.