FabMo/FabMo-Engine #664: Drop-Down DRO covering apps in 1.7.0-rc

tedh-shopbot, 2018-07-30:
May not be a big deal ... but previous DRO was a little more elegant in sharing screen with the home screen and with individual apps.

It might make more sense to leave the I/O display off the small version and only show it for the drop-down version. That would leave more room for visibility of axis locations.

tedh-shopbot, 2018-09-29:
I would just like to elaborate on this one today because it has been bothering me quite a bit.

Two years ago (+?) I felt like we put a reasonable effort into planning for a basic FabMo layout that included the right side, red, DRO. The logic for this was: 1) to preserve a very small amount of the look of the ShopBot software to provide some continuity in appearance and feel; 2) to provide an optional large DRO for those who really liked the machine-look that a DRO provides; 3) to provide a panel with some basic functionality such as a primary location display, i/o interaction, spindle RPM display, useful macro access, location zeroing, etc. that would always be available to apps and developers and that would play well with them in terms of behaving well on the screen. I felt that by 1.6.4 the optional DRO was working well across a wide variety of screens and was a solid spot for further development of the type of functionality we had in mind for it.

I get that we needed to do something different with the keypad, but that is a separate issue. Abandoning the DRO seems very arbitrary to me. The optional right-hand red panel is not wasted space because lots of functionality can be provided there, not the least of which is displaying all 6 axes in a readable and expected location.

The new little pop-up does not achieve any of this kind of functionality. More importantly, it does not play well with other screen displays and is basically unusable on a phone with the Sb4 app, for example. When the pop-up is not open, the 3 rows of data now in the top corner are difficult to read compared to the previous display of just the axis locations.

For me, we are missing the opportunity here for both continuity and clarity, as well as keeping things stable for app developers (though I realize it may just be me who uses the thing, but so be it). And, I can't help but reflect on the aesthetics -- until now the interface has maintained a flat, simple, and color-coordinated look -- I suppose that is an advantage of sticking with a framework. The new little gizmo introduces 3D, adds a new color (there is nothing wrong the use of a spot of absolute color variety in a design for a special purpose, but I'm not sure this rises to that need), and makes use of a novel user-interaction metaphor that is cool, but distracting and non-intuitive.

SW0rden, 2018-10-09:
I totally agree with your last post Ted!  After recently updating to the 1.7.X version, I was verified in my concerns I had some time back, that the interface is going/has gone down a road that I just can't get onboard with.  To be honest, this has gone on for so long, that I really have lost interest in it.  It's what, 4+ years in?  I see little maturity/solidity in the GUI, and not much in the way of urgency to get to a mature product.  As far as thoughts of using it on anything but a desktop unit, I just can't take it seriously. 

Sorry for being painfully honest.

closed 2018-12-06 by tedh-shopbot