FabMo/FabMo-Engine #669: units labels don't track mm default units

scotthz, 2018-09-20:
Engine v1.7.0 on macOS 10.13.6.

- Default units changed to `mm` on Configuration > General.
- Unit labels on Configuration > Machine Envelope remain `in`.
- Unit values on Configuration > Machine Envelope appear to be scaled to `mm`, e.g. displayed `X Max` value was `6` when default units were `in`, but is displayed `152.4` when default units changed to mm. But the unit label is still `in`. 
- Appears to be the same for values under `Manual Control`, but the `Z-Lift` value is `0.5`, which is hard to make sense of.

scotthz, 2018-09-24:
Closing because I can't tell if this is an issue in itself or just one of multiple manifestations of a more pervasive problem. Besides the issue above, I have encountered several others with v1.7.0 which ultimately make its behavior too unpredictable in my environment.

ryansturmer, 2018-09-24:
Reopening this - this is actually an issue, fix in the works.

RobMackie, 2021-03-24:
I just tested this, and as far as I can tell, it is working correctly.  Units and values for machine envelope, for z-lift, and for manual keypad all are work if converting from imperial to metric, and again appear correct if converting after a shift back from metric to imperial.  Also, setting an axis to zero after a move while in metric shows zero on the axis moved in the manual keypad.

I think we can close this as "Unable to reproduce"

closed 2022-10-27 by RobMackie