FabMo/FabMo-Engine #686: Lock-up if stop hit during pause

tedh-shopbot, 2018-11-29:
When running sblogo file from Sb4 App; if you hit stop before motion starts (during pause) tool locks up.
When running sblogo file from Job Manager; if stop hit before motion starts, then Stop Spins , and motions starts, but then you can stop tool.

An issue because of the use of Sb4 interface by New Handibots.

tedh-shopbot, 2019-07-28:
In general, this seems to be an issue when a file is started using the run from app API call. If you hit stop during the spin-up pause; you get the problem. You can now break out, but fault persists into next run of app ... probably something needs to be cleared out in app after a failure.

tedh-shopbot, 2019-11-12:
Stop also fails in situations like the Warm-Up Macro. Then tool is locked up.

kyle-kingsbury, 2022-03-29:
Fixed with PR https://github.com/FabMo/FabMo-Engine/pull/825

closed 2022-03-29 by kyle-kingsbury