FabMo/FabMo-Engine #690: Parsing math in SBP Commands

tedh-shopbot, 2019-11-07:
DOES NOT WORK                              DOES WORK
    JZ, -1 * 1                                            JZ, -1
                                                              JZ, (-1) * 1
    JZ, 1 * -1                                            JZ, (1) * -1
    JZ, 1 * (-1)                                          JZ, (1 * (-1))
                                                               JZ, (1 * -1)

Essentially you need to open the expression with a parentheses in order to have it evaluate and work as an expression. Should be a more general evaluation ...

kyle-kingsbury, 2022-03-21:
I did some test operations and everything seems to work fine. I tried addition, subtraction, multiplication and division. Even dividing by zero just causes the program to exit. No longer an issue?

kyle-kingsbury, 2022-03-22:
Not reproducable. With/without parentheses, added whitespace, no whitespace, seems to have been fixed somehow.

closed 2022-03-22 by kyle-kingsbury