FabMo/FabMo-Engine #758: Motion on Variables Set during runtime Fails with Error

jkeppeler-parthenon, 2021-11-16:
If a Variable is set during runtime (via pause input functionality or potentially if a probe result is stored in a variable during runtime then used in the same job) FabMo will crash with an error "object v does not have attribute toFixed()" in SBPRuntime.prototype.emit_move

Transformations are pre-calculated and thus the lookup for the v object in the transform map can fail if this motion variable is defined after transforms are calculated.

jkeppeler-parthenon, 2021-12-03:
root cause is value returned being saved to config as string not float then being retrieved as a string before calling toFixed() which is a float function.

Corrected in PR #763 

For emit move chain this has been corrected by explicit cast to float via parseFloat() 

toFixed is used over a hundred other times in the codebase (from term search) and with anonymous functions it's not clear how many of these might get a string value from config though most seem to explicitly cast as well or utilize mathematical operators that would coerce the value to a number.

RobMackie, 2021-12-14:
Needs testing. Appears to be fixed by merge of PR #763

RobMackie, 2021-12-20:
Tested with this file:

SA
&myvar = 1
pause, "Change value of &myvar to:", &myvar
mx, &myvar
end

Procedure:
Ran file and when dialog popped up, I entered 5 into the dialog. After file exited, I confirmed via DRO that the spindle had moved to x == 5.
Ran the file again, this time entering the value 0 into the dialog. Afer the file exited, I confirmed via DRO that the spindle had moved to x == 0

tested with version fabmo: 3.0.85
g2core: 101.57.28

RobMackie, 2021-12-20:
Sovled after Pull Request #763

closed 2021-12-20 by RobMackie