FabMo/FabMo-Engine #761: C3 macro hangs fabmo if any prox switch is activated when c3 is started

RobMackie, 2021-11-30:
Problem: 
C3 macro goes to the infinite spinning STOP footer if started with a prox switch activated.

Expected behavior: back off all switches and then run macro

Work Arround: go to editor (macro or file or job manager) and use the "escape" key on keyboard 

question ?? is there an escape key analog on a phone or tablet?

tedh-shopbot, 2021-12-01:
Not able to duplicate this one.

Make sure you are in current master branch, 3.0.85, and have installed the dev-tests profile that accompanies it.
This profile is set up for a DT ... you might have to make a few changes in Macro201 for DT-Max. FYI, the definitive up-to-date macro set (that only exists for the DT) ... is in:   fabmo-profile-dt   on Github.

The expected behavior is that the tool first makes a move in X and Y to make sure it is not parked behind a prox switch. Then it checks X and Y prox switches and attempts to move off any that might be triggered. If unsuccessful it bails with a message (message allows a resume, but user needs to quit of course). This all seems to be working in a quick test just now. I may be missing something in a different case ...

In my experience to date, the failure you report happens when an ON switch is passed into the probe routine as the input to monitor. In this case G2 will report the error that we are not exiting well from at the moment (probably need to make sure we have the probe failure/alarm on Josh's list, just in case). However, there is basically a double process described above that should prevent this ever happening with the current macro.

In theory, we should probably clear all of the switch designations for "action" under inputs in config manager. However, I have tested these by putting random settings in and during a probe, g2 seems to appropriately ignore any other setting for the input during the probe. Also the actions are only called on a change, so some other action switch already being triggered at the start of the macro should have no effect other than in the "interlock" case -- which we have not hooked up yet. 

Maybe you can provide some more detail?

RobMackie, 2021-12-20:
Not reproducible. Suspect it was originally found on a system where the macros had been edited after installation. junked the issue report.

closed 2021-12-20 by RobMackie