FabMo/FabMo-Engine #766: gcode file wthout a feed rate hangs

RobMackie, 2021-12-14:
when this g-code file did not have a speed setting (feed rate set F100 or g1 ... F1), it failed to run in either the editor or a file (g2 does report error of the absence of a speed value). The error is not reported to the user, it is just seen as a lockup, with or without a footer (though it can be escaped). If the user does something that generates some axis motion (say a keypad move), then a default speed is saved for g-code (good?) and this error does not happen. {See previous notes for dev-test 2 on handling errors; this is just a reconfirmation of behavior.}

RobMackie, 2022-01-04:
Let's handle it by popping up a dialog when we get the error from G2 that says there is no feedrate.

kyle-kingsbury, 2022-01-26:
Issue can be replicated by setting feed rate to 0.

kyle-kingsbury, 2022-02-10:
This issue was reported as a gcode issue. My solution produces g2 error reporting in gcode only. In the future, we should implement g2 reporting for all runtimes. This is not a perfect solution. If a "no feed rate" error is received then the error is handled as expected. The code does not produce axial movement and the error is displayed to the user. A different error (such as syntax/formatting) may still produce axial movement while the error is displayed to the screen. I will be writing this behavior up as a separate bug. 

This is the code that I used to test,
g20
g90
g0 x0 y0 z0.5
s1000 m3
g01 f300
g01 z0
g01 x0.5
g01 y1
g01 x1
g01 y1
g01 x1.5
g01 y1.5
g01 x2
g01 y2
g01 x2.5
g01 y2.5
g01 x3
g01 y3
g01 x3.5
g01 y3.5
g01 x4
g01 y4
g01 x4.5
g01 y4.5
g01 x5
g01 y5
g01 x5.5
g01 y5.5
g01 x6
g01 y6
g01 x6.5
g01 y6.5
g0 x0 y0 z0.5
m5
m30

Executed directly 5x, 6th time the program exited prematurely:
info: Clearing the quit pending state. [g2]
info: Resolving promise 9 because of state 4 which is one of 4 [g2]
debug: Finishing up the cycle context. [g2]
debug: call final lines from machine [machine]
debug: call MPO from machine [machine]
info: Disconnected G-Code Runtime [gcode]

Executed directly with stop and resume (1-3 times per run) 6x
7th time program exited prematurely with same log as above

Execute directly with stop and quit 3x
4th time program exited prematurely with same log, this is because of the quit pending state perhaps. 
Execute directly with stop and quit 1x
Then same error as above

I then switched to master branch to see if this is an existing bug, which it is. Seems to do with the quit_pending variable not being set to false in certain situations. Will look into and write up the issue later if it is not already. 

Executed directly 3x with error reporting by switching the f300 to f0. Pressed close in the dialog box each time.
then executed directly with feed rate set back to 300 and everything ran as expected. 

Ran devcheck 1, executed directly and from job manager, stopping and resuming.
Success

Ran devCheck 2 directly and from job manager. Error reported if test_dist is undefined
No error when it is defined
Success

Ran devCheck 3 directly and from job manager. 
Success

Ran devCheck 4, seems confusing. Using the stock code it produces "an error occured" in a dialog box and "Error: Variable $SB_VARTABLEUNITS was used but not defined" in the log. Joshs code from the shared dev check report also produces error
debug: TOCK (Parse file) no tick! [sbp]
debug: _setupTransforms [sbp]
debug: Transforms configured... [sbp]
error: Error: Duplicate label.
I tried this process a couple times and it led to completely different situations where I had to run dev to get things correct again. I must be doing something wrong.
Fail?

Ran devCheck 5 directly and from job manager. Looping working correctly with sub programs but A axis does not function (known g2 bug).
Fail

Ran devCheck 6 directly and from job manager. Seemed fine. moved to x0, displayed message with keypad, I moved with the keypad, then it moved to x0 after closing keypad.
Success

kyle-kingsbury, 2022-02-17:
More work needs to be done here. After the error is reported to the user, the next program execution (sometimes) ends prematurely.

kyle-kingsbury, 2022-02-17:
I made another commit which fixes the footer disappearing. Seems like code got pushed into master that was not merged or something. I believe the program ending prematurely (sometimes) is an existing bug. Will write it up.