FabMo/FabMo-Engine #767: Running file, stop resume, stop, followed by "quit" results in hang

RobMackie, 2021-12-14:
m3
g01 z-0.5 f60
g01 x10 y2 f60
g01 x0 y0
g01 x2 y10
g01 x0 y0
m5
- This worked fine for me too. EXCEPT if you click Quit after just hitting Resume ... now you have the two spinning arrows.
Implicit hit pause, hit resume then  hit Quit and expect a hang.

jkeppeler-parthenon, 2021-12-16:
I believe this is only a partial duplicate of #765 and the PR above does fix part of this.  In specific the hang on quit.

When running the above code after the fix in PR #768 it will run to completion successfully back to back as well as if stop then quit is used.  If stop then resume is used it will hang on resume (spinning circle on resume) until quit is hit then it will successfully quit.  From this quit I am able to run the job again to completion or quit succesfully.  This is all running the above through the editor.

On resume we successfully send the resume through the websocket then emit a ~ to g2 which we receive no response or motion on the g2core appearing to remain in feedhold.  If ~ is still the correct GCode to resume from feedhold (! to initiate ~ to release is currently in use in fabmo) then it appears G2Core is not correctly picking that up while in feedhold.

RobMackie, 2022-01-04:
Reading the notes here, it appears we need to characterize this in sufficient detail to submit it to the g2 team to fix.

jkeppeler-parthenon, 2022-01-04:
I have determined this issue only occurs in the gcode runtime which is very odd given the communications with G2Core are nearly identical from what I see.

Two test cases:
Test A) In editor, Language is gcode run from editor the following:
```
m3
g01 x1 y1 f60
g01 x0 y0
m5
```

During motion press the Fabmo software stop button in the UI.  This correctly enters feedhold.  On pressing resume the resume button is disabled and the spinner UI element is activated to indicate the command is in process.  No action on the motor no further UI updates, light strip remains blinking as in feedhold.  Pressing quit at this point will correctly end the job and allow restart without issue, job will run to completion on restart or can be started then stopped -> quit successfully.

Relevant Logs:

> g2: <-S-1641321490821----- {"sr":{"posx":0.7123,"posy":0.43406}} [g2]
> debug: This Command = pause [websocket]
> debug: Sending a feedhold [g2]
> g2: --S-1641321491001----> ! [g2]
> debug: This Command = pause [websocket]
> debug: Sending a feedhold [g2]
> g2: --S-1641321491054----> ! [g2]
> g2: <-S-1641321491073----- {"sr":{"posx":0.72326,"posy":0.45561,"vel":0.01,"stat":6,"hold":4}} [g2]
> g2: <-S-1641321491079----- {"sr":{"posx":0,"posy":0,"vel":0,"hold":7,"line":24}} [g2]
> g2: <-S-1641321491112----- {"sr":{"posx":0.72326,"posy":0.45561,"hold":0,"line":22}} [g2]
> debug: This Command = resume [websocket]
> info: Creating promise 17 [g2]
> g2: --S-1641321493466----> ~ [g2]

From the log we see normal motion in progress.  Pause continues to send from the UI triggering a send to G2Core of the ! feedhold character until we get the stat 6 (two sends in this case)  on Resume a single '~' is sent and we get no response from g2.  Same behavior is seen when only x axis motion is input. 

Test B) In Editor Language set to OpenSBP, run from editor the following
```
MX, 0
MX, 1
MX, 0
```

During motion press the Fabmo software stop button in the UI.  This correctly enters feedhold.  On pressing resume the resume button is disabled and the spinner UI element is activated to indicate the command is in process.  Moments later motion resumes and runs to completion or through Additional stop -> resume/quit UI use.

Relevant Logs: 

> g2: <-S-1641321872466----- {"sr":{"posx":0.70784,"posy":125.38578}} [g2]
> debug: This Command = pause [websocket]
> debug: Sending a feedhold [g2]
> g2: --S-1641321872558----> ! [g2]
> debug: This Command = pause [websocket]
> debug: Sending a feedhold [g2]
> g2: --S-1641321872607----> ! [g2]
> g2: <-S-1641321872718----- {"sr":{"posx":0.70951,"posy":124.67533,"vel":123.82,"stat":6,"hold":4}} [g2]
> debug: onSTAT ...6 [sbp]
> debug: This Command = pause [websocket]
> debug: Sending a feedhold [g2]
> g2: --S-1641321872804----> ! [g2]
> g2: <-S-1641321872952----- {"sr":{"posx":0,"posy":0,"vel":0,"hold":7,"line":22}} [g2]
> g2: <-S-1641321872984----- {"sr":{"posx":0.70988,"posy":124.51753,"hold":0,"line":21}} [g2]
> info: No wifi, currently in AP mode [network]
> debug: This Command = resume [websocket]
> info: Creating promise 3 [g2]
> g2: --S-1641321885186----> ~ [g2]
> g2: <-S-1641321885222----- {"sr":{"stat":5}} [g2]

From the log we see normal motion in progress.  Pause continues to send from the UI triggering a send to G2Core of the ! feedhold character until we get the stat 6 (two sends in this case and a third after the stat 6 in this log)  on Resume a single '~' is sent and we get a stat 5 from g2.  This is followed by resumption of the movement in the log to match observed behavior.

Full log attached.  These logs include all G2Core communications including prepended set up lines and all responses up until resumption of normal motion.
[767-log.txt](https://github.com/FabMo/FabMo-Engine/files/7810201/767-log.txt)

jkeppeler-parthenon, 2022-01-04:
Further tests in gcode verify that this stall only happens when we first send the m3 to g2core.  If that is omitted:
```
g01 x1 y1 f60
g01 x0 y0 
m5
```
resume works correctly.  This explains the difference between the OpenSBP performance and gcode.
 If we include an SO, 1, 1 in the OpenSBP file to trigger the spindle on M3 we get teh same non response on resume from feedhold.

RobMackie, 2022-01-07:
This problem has been diagnosed as a G2 core problem. Notes are here. https://github.com/FabMo/FabMo-G2-Core/issues/29 
A new g2core.bin image will checked in here and this will be closed.

RobMackie, 2022-01-07:
A new version of g2core.bin.101.57.30 has been correctly built and placed in this repo. The file can be downloaded from github and installed on the controller card to solve this issue. Builds from linux containers prior to the resolution of https://github.com/FabMo/FabMo-G2-Core/issues/29 will result in having this problem. Microchip Studio based builds and and builds with docker containers obtained after the resolution of https://github.com/FabMo/FabMo-G2-Core/issues/29 will not have this problem.

closed 2022-01-07 by RobMackie