FabMo/FabMo-Engine #769: openSBP runtime: run after quit stops early

jkeppeler-parthenon, 2021-12-17:
Can be reproduced  by running the following from the editor:
MX, 0
MX, 2
MX, 0
PAUSE, "EOF" 

Executes correctly if allowed to run to completion by selecting resume form final pause, or if quit from final pause.
If stop is engaged followed by quit to halt the run the next run will execute normally until the next stat 3 is received at which point the run will end.

On execution of the g2.js quit() function queueflush is called and passed a callback.  This callback is saved  as flushcallback.  The g2 handle status report will run this callback on the next receipt of a stat 3.  The callback sent from the quit function includes a write('\x04') kill command.  This means that if something triggers the G2.quit() and the queueflush is not resolved by a stat 3 the callback with the kill remains and will be applied on the next stat three during the next job.

flushcallback is defined by queueflush and only cleared when run by getting a stat three.  This seems to be to ensure after a queueflush that the callback will run.  But if it does not due to the run ending before the queueflush completes this error will occur.  

Need to determine if we should clear the flushcallback as part of creating a cycle context and accept that the callback may not be called in some situations.  Alternatives would include ensuring that the callback is run and cleared also on a stat 4 (END) perhaps but then we're sending a kill into a stat 4 state.  Lastly the kill is also sent as part of the queueflush process making the additional kill in the callback potentially superfluous, it is also possible neither function should send a kill to G2 and this should be handled synchronously instead of via async functions and callbacks and/or with a M30 instead of a kill.

RobMackie, 2021-12-20:
issue #773 duplicates this bug. When this bug is resolved, testing should include the cases described in #773

jkeppeler-parthenon, 2022-01-28:
corrected in PR #788

closed 2022-02-01 by RobMackie