FabMo/FabMo-Engine #770: Keypad UI misbehavior with location box selection ...

tedh-shopbot, 2021-12-19:
This is a little fussy ... but distracting.

When the keypad is just opened, before making a move, a click on the axis-location-input-box selects the box and activates 2 buttons over what had been the arrow keys of the keypad -- one button for "move to" and one button for "set to".

A user can put a new location into one or more of the the axis-boxes and then click the button to move there or to change the current location to the new number (via a change to the g55 offset for the axis or axes).

This works correctly is the keypad has just been opened. BUT IF the user has MADE A MOVE USING A KEY ON THE KEYPAD, then after the click is made to select the axis, any move of the MOUSE (in the area where the "Fixed" slider had previously been) will UNEXPECTEDLY HIDE THE BUTTONS.

 *Can be made more obvious by activating "Fixed" and making a couple of moves in fixed mode before clicking on an axis location.

tedh-shopbot, 2022-03-05:
Ahh ... here is a simpler way to produce issue. Open keypad. Click on location DRO to provide a new GOTO location ... be very careful not to move mouse until after you release button. This WORKS.
Now, click on location DRO and slide mouse slightly before releasing ... the large Green GOTO BUTTON on the right will be lost. Need to be more selective in the button remove behavior. (Clicking on the DRO will again bring up button to override the error.)

kyle-kingsbury, 2022-03-11:
While testing this I realized it is the product of receiving two stat 3s in a row from g2. This is a command to "go to" x10 from x 0. The only lines of interest are the last two, the rest is just for reference.

info: Arming the machine for runtimeCode [machine]
info: Firing automatically since authorization is disabled. [machine]
info: Machine is deauthorized. [machine]
info: CODE!! [manual]
info: {"cmd":"goto","move":{"X":10,"Y":0.001,"Z":0.02,"A":0}} [manual]
info: Priming driver (manually) [g2]
g2: --S-1647028584205----> G90\nG0 X10 Y0.001 Z0.02 A0\nM0\nG91 [g2]
g2: <-S-1647028584209----- {"r":{},"f":[1,0,4]} [g2]
g2: <-S-1647028584210----- {"r":{},"f":[1,0,23]} [g2]
g2: <-S-1647028584211----- {"r":{},"f":[1,0,3]} [g2]
g2: <-S-1647028584212----- {"r":{},"f":[1,0,4]} [g2]
g2: <-S-1647028584461----- {"sr":{"posx":0.27635,"vel":224.69,"stat":5}} [g2]
g2: <-S-1647028584712----- {"sr":{"posx":1.67646,"vel":360}} [g2]
g2: <-S-1647028584963----- {"sr":{"posx":3.18825}} [g2]
g2: <-S-1647028585214----- {"sr":{"posx":4.70004}} [g2]
g2: <-S-1647028585465----- {"sr":{"posx":6.22383}} [g2]
g2: <-S-1647028585716----- {"sr":{"posx":7.73562}} [g2]
g2: <-S-1647028585968----- {"sr":{"posx":9.24186,"vel":346.49}} [g2]
g2: <-S-1647028586219----- {"sr":{"posx":9.99561,"vel":15.83}} [g2]

g2: <-S-1647028586289----- {"sr":{"posx":10,"vel":0,"stat":3}} [g2] <----first stat 3

g2: <-S-1647028588538----- {"sr":{"stat":3}} [g2] <----second stat 3

When the "go to" and "set" buttons disappear it is a product of the second stat 3 resetting the keypad. If I wait until the second stat 3 has passed, I can no longer replicate the issue. Not sure why two stat 3's would be needed. Also seems strange because I thought G2 only produces status reports on fields that have changed.

tedh-shopbot, 2022-03-11:
@kyle-kingsbury Wow. That is interesting. I was thinking it was something related jquery being set to the wrong event on the dashboard side of things.

kyle-kingsbury, 2022-03-16:
To reliably trigger this bug I click any direction on the manual keypad, then click on one of the numeric axes coordinates to trigger the "goto" and "set" buttons. Once the next stat comes in the buttons disappear. My fix adds a flag that is triggered when one of the axes is selected which is when the "goto/set" buttons appear. It is then set to false in a couple places as needed.

Each test that I ran I waited for the next stat to come in and see the outcome.

- Jog x,y,z,a (not "fixed")
- adjust movement speed then Jog x,y,z,a (not "fixed")
- zero all axes using "zero" button (not "fixed")
- Jog x,y,z,a ("fixed" mode)
- adjust movement speed then Jog x,y,z,a (fixed" mode)
- zero all axes using "zero" button ("fixed" mode)
- Make a "fixed" keypad move using UI and the arrow key then switch to "goto/set" mode (usual method to trigger bug)
- - Make a not "fixed" keypad move using UI and the arrow key then switch to "goto/set" mode (usual method to trigger bug)

kyle-kingsbury, 2022-03-21:
Closed with PR https://github.com/FabMo/FabMo-Engine/pull/834.

closed 2022-03-21 by kyle-kingsbury