FabMo/FabMo-Engine #771: Spindle stops prematurely during shopbot logo file cut

RobMackie, 2021-12-20:
In the DEV CHECKS app, the first test is to run the ShopBot Logo File from the editor.

When the file is run, the spindle enable output #1 lights up, but then within 2 or 3 seconds goes out. This is bad. Resuming a cut with the spindle off will break bits.

Looking at the trace, you can see where some part of the opensbp interpretation to gcode results in emitting an m0.

The following two pieces of gcode (runnable from the gcode editor) show why this is a bad idea. (watch the DRO motion in the X axis and the value of output 1 for each of the following two files:
good file:
g90
g20
m3
g1 x5 f50
g1 x0 
m5
m30

bad file:
g90
g20
m3
g1 x5 f50
m0
g1 x0 
m5
m30

This makes it clear that we should not be using m0 mid job if we expect the spindle to keep spinning.

A good fix will alter the openSBP run time so it doesn't use M0 in the middle of a file, while maintaining all pause and dwell behaviors that the openSBP runtime currently implements.

kyle-kingsbury, 2021-12-27:
M0 is an unconditional stop. It should stop the machine, in whatever position it is in, and the spindle would continue running (if it is running before the M0). The M0 gcode is not being interpreted correctly.

tedh-shopbot, 2021-12-28:
Yes. I figured we could just remove those lines. 

Earlier, when trying to get G2's attention before any movement had started in a file, we were using those M0's to indicate to G2 that a cycle had started and that it should attend to non-motion instructions (probably to get it from stat:3 to stat:4). At the time, we needed it to turn outputs on and at that earlier time we were trying to use the g-code dwell to impose a PAUSE (now implemented on the FabMo side and not in g-code) which depended on the M0. 

I think we have basically fixed all that, so it would be worth running a few files with outputs and other pre-motion commands just to make sure.

However, as you point out, M0 should not turn off the spindle and at some point we should fix this in G2.

"Stack-breaking" is now stated as "motion-segment breaking" and an "OpenSBP stack" is considered a "motion segment".

tedh-shopbot, 2021-12-28:
There's a philosophical reason for needing to consider a chunk of motion as a segment that has a distinct start and stop -- another day. :)

kyle-kingsbury, 2021-12-28:
Thanks for the response. That helps a lot. I will do some testing and report back!

kyle-kingsbury, 2021-12-28:
After commenting out the two lines I ran the following blocks of code.
One in absolute with pauses:
SO 1,1
SA
PAUSE 5
PAUSE "PREMOVE ACTIONS"
MX, 2
MY, 3
MX, 0
MY, 0
END

One in absolute without pauses:
SO 1,1
SA
MX, 2
MY, 3
MX, 0
MY, 0
END

Then ran the same blocks in relative mode. They all ran without error and the spindle stayed on throughout the runtime. 

I then tried the source code as it was originally (with the stack breaking M0's). The spindle shuts off after the SR. Same if ran with an SA.
SO 1,1
SR
PAUSE 5
PAUSE "PREMOVE ACTIONS"
MX, 2
MY, 3
MX, 1
MY, 1
END

If you remove the pauses the movement is very buggy. It moves to the final coordinate, then moves back to the origin, then shuts the spindle off after it starts running. Same if ran in absolute. This proves that removing the stack breaking M0's is a necessary fix.   
SO 1,1
SR
MX, 2
MY, 3
MX, 1
MY, 1
END

tedh-shopbot, 2021-12-28:
Sounds like we're good. =ted

closed 2021-12-29 by RobMackie