FabMo/FabMo-Engine #773: Running shopbot logo file from editor, Stop/Quit, re-run shopbot logo file again - fails to cut

RobMackie, 2021-12-20:
Doing Dev Checks step 1. Running shopbot logo file from editor

Runs fine. 
Stop/resume multiple times is fine
Stop/quit works, BUT if you try to run the file a second time, the file starts but never gets to moving the cutting head, and silently ends prematurely.

A good fix for this would leave the machine in a state where it could run a "next file" after a "last file" was run through "stop/quit"

Here is a much simpler version of the same problem.
SA
m2, 0,0
mx, 5
'test pause
pause
my, 5
m2, 0,0
end

hit stop/quit on a first run, then run it a second time. the pause dialog never displays and the file is left aat x=5, y=0. 

* running the file again works - it recovers by the 2nd attempt to rerun the file.
** quitting from the pause dialog does not cause the problem.  Quitting from the footer does cause the problem. Again, the problem isn't visible until the run of the file that immediately follows the run that was subjected to stop/quit. And after the run that fails, future runs will succeed, until another stop/quit is performed. Then the very next iteration will fail again.

RobMackie, 2021-12-20:
This turns out to be a duplicate of the more completely analyzed issue #769 
This case should be tested after a fix to #769 but the details on that issue are  more useful for debugging.

jkeppeler-parthenon, 2022-01-28:
corrected in PR #788

RobMackie, 2022-02-01:
Closeed with PR #788

closed 2022-02-01 by RobMackie